uholeschak/ediabaslib #2: Paths/filenames for Elm327 replacement firmware need fixing

grnassar, 2017-08-02:
In the "Replacement Firmware for ELM327" page, in "Programming of the Processor:"

`The source for the firmware could be found in the subdirectory CanAdapterElm. In this subdirectory there is also a flash tool (subdirectory UpdateLoader) that allows firmware updates using the bootstrap loader once the firmware is programed.  The latest firmware version will be always included in Deep OBD for BMW and VAG.`

In the current binary release, there are three folders in the CanAdapterElm subdirectory: bc04, hc04, and default.  There is no UpdateLoader folder -- that is in the CanAdapter folder, not CanAdapterElm.  No guidance is given for what the difference is for the bc04, hc04 and default folders.  Also no clarity as to the "unified" and "non-unified" within those folders.

The files in the CanAdapter folder match up much better to the descriptions on the firmware page, but they are also over a year older (and of course don't match the subfolder name in the instructions), so I imagine they are not the ones to use?  But one can't really be certain with the instructions as they are, I think.

uholeschak, 2017-08-05:
Should be fixed now.

closed 2017-08-05 by uholeschak

grnassar, 2017-08-16:
I'm sorry, but it seems like the documentation is again already out of date.  The documentation names hex files that are already there, and two of the three directories that are supposed to be in CanAdapterElm aren't either.  In fact, I can no longer find hex files in the repository at all.