uholeschak/ediabaslib #5: no connection to interface

harryberlin, 2017-09-16:
Hi

I did replace the api32.dll in ediabs/bin path with yours.
Needed to install vcredist 2015 runtimes.
Paired the bluetooth dongle and got com6 and com7.
Edited obd.ini to com6.
Inpa is loading.
Battery and Ignotion stays off.
After selecting my e39 and the lcm i get the error about an interface error.
The same with an usual usb interface.
Windows 7 Pro 32bit

How to get it working?

uholeschak, 2017-09-17:
Hello,
you did not follow the description in:
https://github.com/uholeschak/ediabaslib/blob/master/docs/Replacement_firmware_for_ELM327.md#use-the-adapter-with-inpa-tool32-or-ista-d
You have to use the EdiabasLibConfigTool.exe, but before you have to undone any manual changes!

harryberlin, 2017-09-17:
that means, if i'd like to use inpa, i have to change the firmware? and for deepobd i have to switch back to other firmware?

uholeschak, 2017-09-17:
Maybe I don't really understand your problem.
Are you using a Deep OBD Bluetooth dongle or something else?
The description is only working with a Deep OBD Bluetooth dongle
For USB you have to edit the EdiabasLib.config file not the obd.ini.

harryberlin, 2017-09-17:
i have an deep obd bluetooth dongle and like to use it with inpa.
which file are required to copy in ediabas path and which destination exactly?

uholeschak, 2017-09-17:
So why you don't follow the description and copy nothing manually?

harryberlin, 2017-09-17:
OK, i try it with the config tool.
Do i have to backup the current ediabas path, to switch back to usual usb obd interface, later?

uholeschak, 2017-09-17:
There is a unpatch button in the tool.

closed 2018-01-26 by uholeschak