uholeschak/ediabaslib #6: Jobs with some empty arguments don't work properly

terraphantm, 2017-10-29:
I've noticed that EdiabasLib doesn't handle jobs where some of the arguments are left empty. Below I've attached examples of the Ediabas vs EdiabasLib behavior in tool32:

EdiabasLib
```

apiJob("08MSV70","diagnose_mode","default;PC115200","")

Satz : 0
  VARIANTE                        = 08MSV70
  OBJECT                          = 08msv70
  JOBNAME                         = DIAGNOSE_MODE
  SAETZE                          = 1
  JOBSTATUS                       = 
  UBATTCURRENT                    = -1
  UBATTHISTORY                    = -1
  IGNITIONCURRENT                 = -1
  IGNITIONHISTORY                 = -1
Satz : 1
  JOB_STATUS                      = OKAY
  _TEL_AUFTRAG                    = 7 Bytes
    0000 : B8 12 F1 03 10 81 05                               ¸.ñ...
  _TEL_ANTWORT                    = 8 Bytes
    0000 : B8 F1 12 03 50 81 05 8C                            ¸ñ..P.?  

apiJob("08MSV70","set_parameter",";115200","")

EDIABAS Error 16 : IFH-0006: COMMAND NOT ACCEPTED

apiJob("08MSV70","set_parameter",";115200;;15","")

EDIABAS Error 16 : IFH-0006: COMMAND NOT ACCEPTED

```

Ediabas 7.3.0

```
apiJob("08MSV70","diagnose_mode","default;PC115200","")

Satz : 0
  OBJECT                          = 08msv70
  SAETZE                          = 1
  JOBNAME                         = diagnose_mode
  VARIANTE                        = 08MSV70
  JOBSTATUS                       = 
  UBATTCURRENT                    = -1
  UBATTHISTORY                    = -1
  IGNITIONCURRENT                 = -1
  IGNITIONHISTORY                 = -1
Satz : 1
  JOB_STATUS                      = OKAY
  _TEL_AUFTRAG                    = 7 Bytes
    0000 : B8 12 F1 03 10 81 05                               ¸.ñ...
  _TEL_ANTWORT                    = 8 Bytes
    0000 : B8 F1 12 03 50 81 05 8C                            ¸ñ..P.?  

apiJob("08MSV70","set_parameter",";115200","")

Satz : 0
  OBJECT                          = 08msv70
  SAETZE                          = 1
  JOBNAME                         = set_parameter
  VARIANTE                        = 08MSV70
  JOBSTATUS                       = 
  UBATTCURRENT                    = -1
  UBATTHISTORY                    = -1
  IGNITIONCURRENT                 = -1
  IGNITIONHISTORY                 = -1
Satz : 1
  JOB_STATUS                      = OKAY

apiJob("08MSV70","set_parameter",";115200;;15","")

Satz : 0
  OBJECT                          = 08msv70
  SAETZE                          = 1
  JOBNAME                         = set_parameter
  VARIANTE                        = 08MSV70
  JOBSTATUS                       = 
  UBATTCURRENT                    = -1
  UBATTHISTORY                    = -1
  IGNITIONCURRENT                 = -1
  IGNITIONHISTORY                 = -1
Satz : 1
  JOB_STATUS                      = OKAY
```

uholeschak, 2017-10-30:
Thanks, I will check that, I expect that the empty argument is simply removed.

uholeschak, 2017-10-30:
The problem are not the parameters but the command itself. set_parameter tries to set parameter for the EDIC interface, what is not possible with EdiabasLib. I have changed the error code to "concept not implemented".

closed 2017-10-30 by uholeschak

terraphantm, 2017-10-30:
I'm not using an EDIC interface (using the same OBD interface in both instances), and communications stop working if I raise the module's baudrate without also sending the "set_parameter" command (which I can't do in EdiabasLib). 

As far as I can tell, it *checks* if the interface is an Edic interface so that it can send stop command to the Edic.

uholeschak, 2017-10-30:
Hello,
I don't have the 08MSV70.prg file, so I used flash.prg fro testing. Could you send the file to ulrich@holeschak.de?
I may case set_paramter tries to set the communication paramater for concept 0, which is EDIC, this is rejected.

uholeschak, 2017-10-30:
Fixed pars handling for empty arguments.