uholeschak/ediabaslib #7: ELM327 Firmware Device id

santinio, 2017-12-30:
ELM327 comes with a PIC18F2480 but the provided Microchip project is configured for a PIC18F25K80.

uholeschak, 2017-12-30:
That's not correct, all OBD EM327 adapters i have seen use the PIC18F25K80 chip

closed 2017-12-30 by uholeschak

santinio, 2017-12-30:
Please check https://en.wikipedia.org/wiki/ELM327.

[image]

uholeschak, 2017-12-30:
Then you should correct that wiki text ...

santinio, 2017-12-30:
A PIC18F2480 is also soldered on my ELM327.

uholeschak, 2017-12-30:
Then you could adjust the settings locally for yourself if you like ...

uholeschak, 2017-12-30:
I have now seen that there are two types:
A standard one (ELM327 PIC18F2480) and a low voltage type (ELM327L PIC18F25K80):
https://www.elmelectronics.com/products/dsheets/
The OBD adapters normally only use the low voltage type.

XantheFIN, 2021-02-09:
Chinese when cracked the ELM327 figured out to use code on PIC18F25K80 which was cheaper IC to put code on. There exist even forum thread on chinese forums when someone introduced it.

uholeschak, 2021-02-09:
These are not real PIC processors only some other chips with a fake label.

XantheFIN, 2021-02-09:
> These are not real PIC processors only some other chips with a fake label.

Hold your horses. Don't confuse irrelevant things. Please use caution when reading as i am specific here. I wasn't referring to those black unknown glue IC ELM chips there is as i mentioned chinese discussion of what i referred at from back of time when they did make that move to PIC18F25K80 instead of what ELM was originally made for. I am not talking what is real pic chip or not... what a hell. I am telling why there is even 25K80 PIC chosen in chinese clones. It has nothing to do being low voltage or not.

Here is pieces from announcement:

"The original ELM327 uses PIC18F2480 chip scheme, which is more expensive. If you do with PIC18F25K80, the cost can be reduced

--

The specific work of the scheme is to port the code from PIC18F2480 to PIC18F25K80, after analyzing Datasheet, found that the two ICs belong to the same series

--

Tanscoding First of all, recompile 2480, then 25K80 compilation,-- but does. not mean that you can directly use, because from the data, 25K80 register location, some bits of labeling, some register configuration methods are not -- , we selected 2480 to anti-compile just obtained 25K80 code, The resulting ASM and the original 2480 ASM comparison, found that there are dozens of lines of code not --, which shows that the registers involved in this in the two ICs in different bits, we focus on checking the function of these registers in the two ICs, if to, do not need to change, not to manual correction

--

Relying on our strong anti-compilation technology, the project was successfully completed."

And they told to be selling code V1.5 forward in there back then.