uholeschak/ediabaslib #11: Selective extraction of ECU files

jbvo, 2018-02-17:
I'm using DeepODB on a Nexus 5X with 16 GB flash. On this phone in reality 16 GB becomes 14.5GB and of that about 4 GB is used by the Android system. After installing some apps typically only a few GB is left for data. In this situation, it seems a pity that I need to extract the complete 2.3 GB of ECU files.

I was thinking it could be interesting to add an option to only extract the modules that are used in the installed page configs or even only the current page config. What do you think?

uholeschak, 2018-02-17:
The problem is that you simply don't know the dependencies of the ECU files, so it's difficult to detect which files are required. On the other hand memory is very cheap now and will grow in future.

closed 2018-02-17 by uholeschak

uholeschak, 2018-02-17:
But what you could do:
Disable the automatic ECU file check at startup in the global settings, then you could delete some files manually.

rs38, 2020-02-03:
after deleting some big and unused .prg files on the Android Device, I disabled the file check at startup, but it still wants to re-extract the ECU files when entering the Ediabas Tool. Is there sth. like a dirty flag?

uholeschak, 2020-02-03:
I think this disables only the CRC check, but not the file size check. You are right, this will not help you.

uholeschak, 2020-02-04:
I have checked the behavior now:
You have to switch off the ECU check **before** removing the files. If the check fails once a file is deleted to mark data as invalid.