uholeschak/ediabaslib #13: Cannot display basic OBD info

hanynowsky, 2018-04-03:
I replicated your E60 config pages for my E46 N42 (EK924) but only the Error tab is working. The other tabs are empty.
I only want a second tab named: **Monitor** with the following info:
- Battery Voltage
- LTFT Bank1
- LTFT Bank 2
- STFT Bank 1
- STFT Bank2
- RPM
- Coolant Temperature in C°

How can I properly do it?
Thanks for your response.

uholeschak, 2018-04-03:
I think it's easier to use the configuration generator and generate a basic file that contains the required data and then modify it manually.

hanynowsky, 2018-04-03:
@uholeschak  That is what I desperately did before trying to modify your E60 files.
Actually when I use the configuration generator, it generates only the Error page Tab. If I select each ECU , then it generates a page for each Module, but they are empty.
- For example ME9K_NG4.ccpage
```
﻿<?xml version="1.0" encoding="utf-8"?>
<fragment xmlns="http://www.holeschak.de/BmwDeepObd" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://www.holeschak.de/BmwDeepObd BmwDeepObd.xsd">
  <page name="!PAGE_NAME" display-mode="grid" fontsize="small" gauges-portrait="2" gauges-landscape="4" logfile="ME9K_NG4.log">
    <strings>
      <string name="!PAGE_NAME">ME9K_NG4</string>
      <string name="!JOB#AIF_LESEN#AIF_ADRESSE_HIGH">AIF_ADRESSE_HIGH</string>
    </strings>
    <jobs sgbd="me9k_ng4">
      <job name="AIF_LESEN">
        <display name="!JOB#AIF_LESEN#AIF_ADRESSE_HIGH" result="AIF_ADRESSE_HIGH" format="" log_tag="AIF_ADRESSE_HIGH" />
      </job>
    </jobs>
  </page>
</fragment>
```

uholeschak, 2018-04-03:
In your file is at list one line AIF_ADRESSE_HIGH.
What do you mean with empty? You see the label but no value?
The this means you don't get valid data from the car.
You could test reading with test button on the ECU page.

hanynowsky, 2018-04-04:
Thanks for your assistance.