uholeschak/ediabaslib #14: logo design for deep OBD application

harunbadilli, 2018-04-19:
I designed a logo for the deep OBD application. I was inspired by the letter "D". Color options, whether you want a single color or dual color as you want to send the format you want to prepare. I think the new logo of the application will be more noticeable. I hope you like it.

[image]

[image]

[image]

[image]

harunbadilli, 2018-04-19:
The files you want on the link is ready.
https://drive.google.com/open?id=1Gojc9u4ZrU3XCZuRRp3_E4EbsXqQwkBL

uholeschak, 2018-04-20:
Thanks,
basically I have checked the files, but there are some problems.
- EdiabasLibConfigTool AppIcon must be an .ico file, please check the original one which sizes and formats it has to contain.
- The ic_stat_obd files have be use a special format:
https://developer.android.com/guide/practices/ui_guidelines/icon_design_status_bar.html
There is also an generator for these files:
https://romannurik.github.io/AndroidAssetStudio/icons-notification.html
All files could only be used if they follow the specification, format and size exactly.

For general application icons there are these guidelines:
https://material.io/guidelines/style/icons.html#icons-product-icons
There seems to be now a new type of icon:
https://developer.android.com/guide/practices/ui_guidelines/icon_design_adaptive.html
General guidelines:
https://developer.android.com/guide/practices/ui_guidelines/icon_design.html

Regards,
Ulrich

harunbadilli, 2018-05-03:
First of all, I'm sorry I'm late. I made the arrangements and I sent a pull request.  https://github.com/uholeschak/ediabaslib/pull/16

uholeschak, 2018-05-06:
Thanks, should be now all updated.

closed 2018-05-06 by uholeschak