uholeschak/ediabaslib #18: ELM327 BT Connection

colaas, 2018-05-10:
I have an ELM327 Bluetooth adapter that reports version 1.5 but doesn't actually support anything above 1.4 (I checked with ELM327 Identifier). When I connect with the Deep OBD android app, the connection just drops every couple of minutes for a minute or so. Is this the adapter's fault?
Also if I switch to another app and come back it always says there have been errors and wants to send a log file. Should I do it each time?

uholeschak, 2018-05-10:
Write you AppId from the main menu here and send me a trace file, the I could check it.

colaas, 2018-05-10:
The AppId is bcd26f7f-566e-46a1-88aa-e7667a49e6fe , there should already be some traces from weeks ago. I will send new ones as soon as I can.

uholeschak, 2018-05-10:
I found 2 traces, the communication with the ELM seems to be ok, but there is simply no response from the car, also for different ECUs.
It's difficult to guess what's the cause of the problem.

closed 2018-06-13 by uholeschak