uholeschak/ediabaslib #19: DPF meters since regeneration

colaas, 2018-05-10:
I cannot find the value for the meters traveled since the last Diesel Particulate Filter regeneration. I have found exhaust temperature before DPF, pressure and other items related, but this one doesn't seem to be there and I can't understand German well. Can you point me in the right direction? 
Thanks

uholeschak, 2018-05-10:
That's difficult to tell, it depends from your ECU type. You could use the Yandex translation.

closed 2018-05-10 by uholeschak

colaas, 2018-05-10:
I have tried using Yandex but it is blocked after 24 hours. They want my mobile number.
I will try those, thanks.

uholeschak, 2018-05-10:
The Yandex translation is also blocked after a certain amount of translations per time, the you have to wait.

colaas, 2018-05-13:
Thanks! I found it.
Can I ask you what STAT STRECKE BEI ERFOLGREICHER REGENERATION WERT is? It looks like it is 10000km in the future.
Yandex/Google translate are not very clear!

uholeschak, 2018-05-13:
This means the distance after successful regeneration,
I expect its the distance until the next regeneration will occur.

colaas, 2018-05-13:
Is it odometer reading when it will force regeneration if it has not run?
This is the value, car has 169k km:
[image]

tkosig, 2018-05-18:
It returns the odometer reading when the last successful regeneration happened.

uholeschak, 2018-05-18:
I expected that.

colaas, 2018-05-20:
It reads 178036754.233987 but my car has 169000km. Last successful regeneration was 5km ago.

uholeschak, 2018-05-20:
Then it seems that the assumption was incorrect.