uholeschak/ediabaslib #23: ELM327 Wifi wird nicht erkannt

Profex86, 2018-06-20:
Hallo, ich habe einen ELM327 WIFI Adapter von Carly

ip: 192.168.0.10
Port: 35000
Version 1.4b

Wird leider von der Ediabaslib nicht erkannt. Hätte es mit Ista benutzt.

Was muss ich an dem Adapter ändern, damit es funktioniert. lg

uholeschak, 2018-06-20:
Das Konfigtool unterstützt nur die ELM Ersatzfirmware.
Sie müssen EdiabsLib.ini manuell editieren.

closed 2018-06-20 by uholeschak

Profex86, 2018-06-20:
Sie meinen EDIABAS.INI ? Gibt es denn eine Bespieldatei?

uholeschak, 2018-06-20:
Hier gibt es eine Doku für die Parameter:
https://github.com/uholeschak/ediabaslib/blob/master/docs/EdiabasLib.config_file.md

Profex86, 2018-06-20:
Danke das Sie schnell antworten. Ich habe Jetzt die Api32.dll und EdiabsLib.ini in Rheingold\Ediabas\BIN gepackt und 

        <add key="Interface" value="STD:OBD"/>
        <add key="EcuPath" value="E:\BMW\ISTA-D\Rheingold\Ecu"/>
        <add key="RetryComm" value="1"/>
        <add key="TraceBuffering" value="0"/>
        <add key="IfhTraceBuffering" value="1"/>
-->
        <add key="ObdComPort" value="ELM327WIFI"/>
        <add key="AdsComPort" value=""/>
        <add key="LockTrace" value="0"/>
        <add key="EnetRemoteHost" value="auto:all"/>
        <add key="ApiTrace" value="0"/>
        <add key="IfhTrace" value="0"/>

uholeschak, 2018-06-20:
Die ELMs funktionieren nur mit Fahrzeugen die CAN verwenden. Kann also sein, das es bei Ihnen gar nicht geht.
Sie können IfhTrace auf 2 setzen, dann kann man im Trace die Ursache sehen ...

Profex86, 2018-06-20:
Per Handy und dem besagten Adapter funktioniert aber alles (Codieren und DTC)
von Baujahr 97 bis Heute.

IfhTrace habe ich auf 2 gesetzt, aber nirgendswo wird eine Trace Datei erstellt.

ich benutze die Rheingold Version 4.11.32.

Also ich habe NUR Api32.dll und EdiabsLib.ini in Rheingold\Ediabas\BIN 
aus EdiabasLibConfigTool kopiert und angepasst..

Lg

uholeschak, 2018-06-20:
Geht das mit Deep OBD oder nur mit anderen Apps? Mit Deep OBD geht es nur mit CAN.
Das Trace File wird im Trace Verzeichnis unterhalb von von Bin erstellt.

Profex86, 2018-06-20:
Geht mit 

Windows: ScanmasterELM
iPhone: Mit allen Verfügbaren Apps zB Carly (BMWWhat), OBD Doctor, Car Scanner ELM, EOBD, IOBD2 und so weiter...

•Protokolle:
•SAE J1850 PWM (41.6 kbaud)
•SAE J1850 VPW (10.4 kbaud)
•ISO 9141-2 (5 baud init, 10.4 kbaud)
•ISO 14230-4 KWP (5 baud init, 10.4 kbaud)

uholeschak, 2018-06-20:
Das ist zwar nett, hilft aber nicht.

Profex86, 2018-06-20:
LockTrace war die lösung...

executeJob('d_cas.grp'): 'IDENTIFIKATION'
executeJob('d_cas.grp'): 'INITIALISIERUNG'
ELM CMD send: 'ATD'
ELM CMD rec: 'OK

>'
ELM CMD send: 'ATE0'
ELM CMD rec: 'ATE0
OK

>'
ELM CMD send: 'ATSH6F1'
ELM CMD rec: 'OK

>'
ELM CMD send: 'ATCF600'
ELM CMD rec: 'OK

>'
ELM CMD send: 'ATCM700'
ELM CMD rec: 'OK

>'
ELM CMD send: 'ATPBC001'
ELM CMD rec: 'OK

>'
ELM CMD send: 'ATSPB'
ELM CMD rec: 'OK

>'
ELM CMD send: 'ATAT0'
ELM CMD rec: 'OK

>'
ELM CMD send: 'ATSTFF'
ELM CMD rec: 'OK

>'
ELM CMD send: 'ATAL'
ELM CMD rec: 'OK

>'
ELM CMD send: 'ATH1'
ELM CMD rec: 'OK

>'
ELM CMD send: 'ATS0'
ELM CMD rec: 'OK

>'
ELM CMD send: 'ATL0'
ELM CMD rec: 'OK

>'
ELM CMD send: 'ATCSM0'
ELM CMD rec: '?

>'
*** ELM invalid response: '?

>'
ELM optional command 'ATCSM0' failed
ELM CMD send: 'ATCTM5'
ELM CMD rec: '?

>'
*** ELM invalid response: '?
[... 573 lines cut ...]
ELM CAN rec: ''
ELM Data mode aborted
ELM CMD send: 'ATMA'
*** Receive timeout
*** No header received
 (Send): 83 72 F1 22 F1 50 49 
Send SF
ELM send SPACE
ELM data mode not stopped: 
>
ELM CAN send: '720322F150000000'
ELM CAN rec: 'CAN ERROR'
ELM CAN rec: ''
ELM Data mode aborted
ELM CMD send: 'ATMA'
*** Receive timeout
*** No header received
SetError: EDIABAS_IFH_0009
executeJob Exception: Error occured: EDIABAS_IFH_0009
executeIdentJob Exception: executeJob
Stop comm thread
ELM send SPACE
ELM data mode not stopped: 
>
Stop comm thread

uholeschak, 2018-06-20:
Dort steht:
ELM CAN rec: 'CAN ERROR'
Also ein Fahrzeug ohne CAN, geht also nicht.

Profex86, 2018-06-24:
Hallo, beim F10 eben getestet, selber Fehler....
Ich habe eben ein altes Android Telefon ausgegraben um Deepobd zu installieren.. (Aus dem Playstore).. Soweit so gut, nachdem ich das Programm starte, sagt er mir, das er die ECU Daten entpacken muss.. Das ganze dauert dann ca. 10 Minuten und wenn er 100% erreicht hat, dann haut er eine Fehlermeldung raus, das die zip Datei fehlerhaft wäre.... Wo finde ich diese Zip Datei damit ich sie manuell entpacken und auf die SD Karte schieben kann? 

Liebe grüße

uholeschak, 2018-06-24:
Die Datei kann man nicht manuell entpacken sondern ist im obb enthalten.
Verwenden Sie die aktuelle App Version 1.2.113? In der Version davor gab es ein Problem bei bestimmten Spracheinstellungen.

Profex86, 2018-06-24:
Ja, eben nochmal neu aus dem PlayStoe geladen. Version 1.2.113.

Genaue Fehlermeldung;

Fehler
Das Entpacken der Dateien ist fehlgeschlagen.
Die Zip-Datei scheint beschädigt zu sein.

Wie gesagt, läuft erstmal bis 100% durch...

uholeschak, 2018-06-24:
In welcher Phase tritt das auf? Beim Prüfen der Dateien oder noch beim entpacken?
Android Version?
Im Notfall können Sie mir mal ein Video nach ulrich@holeschak.de senden.

Profex86, 2018-06-24:
Frage, die ECU Daten müssen entpackt werden (Größe ca. 2.3GB) ------->Daten vorbereiten------>Archiv vorbereiten-----> Fehler!

Wenn ich die App komplett deinstalliere und neu installiere, dann kommt er bis 100% Entpacken und dann der Fehler...

Android-Version 4.4.4

uholeschak, 2018-06-24:
Also tritt das dann noch beim Archiv vorbereiten auf? Da werden die Daten entschlüsselt.

Profex86, 2018-06-24:
Ja außer wenn ich die App komplett runter schmeiße und alles von vorne beginne, dann kommt er bis 100% entpacken.

uholeschak, 2018-06-24:
Da ist jetzt schwierig zu sagen wo das Problem liegt.
Das Android ist auch schon sehr alt, können Sie das auf einem neueren Gerät probieren?

Profex86, 2018-06-24:
Ja kann ich, ich probiere es gleich mal mit BlueStacks, wenn das nicht klappt, dann probiere ich es auf den Asus Tinker Board, mache da mal später Android zum testen drauf.

Profex86, 2018-06-26:
Stimmt lag am Gerät. Diesen Adapter der angeboten wird, finde ich leider nicht bei ebay. Andere Möglichkeit zum kaufen?

Und vor allem, ist es voll kompatibel zu ista+ bzw Rheingold?
Lg

uholeschak, 2018-06-26:
Wahrscheinlich gibt es den Adapter wieder in eine Woche, Sie müssen sich dann beeilen.
Sie müssen das Konfig Tool verwenden und die Installationen patchen.

Profex86, 2018-06-26:
Gibt es denn irgendwo eine Möglichkeit den Adapter vorzubestellen?

uholeschak, 2018-06-26:
Nein, das wäre für mich zu aufwendig.