uholeschak/ediabaslib #28: New logo/icon proposal

mansya, 2018-11-01:
Good day sir. I am a graphic designer and i am interested in designing a logo for your good project. I will be doing it as a gift for free. I just need your permission first before I begin my design. Hoping for your positive feedback. Thanks

uholeschak, 2018-11-01:
Thanks, if you want you could try.
The current icon has also been designed by somebody else.
Please check the repository for the different image formats and sizes, there are also monochrome variants.
Creating these variants requires most time, for some you have to read the android image specification, there are also android tools to convert the images.
In the final form also vector data is required to allow generation of other sizes.

mansya, 2018-11-03:
hi sir. I already designed the logo for this project. I made the OBD word the real logo. How ? do you like it ? if yes please inform me. and if there are changes please tell me to
[image]
and about vector files and other variant icons. I always provide it. you can check my repo.

uholeschak, 2018-11-04:
Thanks, basically I like it, but I will ask some other people in the next days first ...

mansya, 2018-11-04:
can I continue to pull requests? if yes. I will give all files from this icon.

uholeschak, 2018-11-04:
I think it's better if you wait, maybe some changes are required.

mansya, 2018-11-04:
ok sir. I will wait for him.

mansya, 2018-11-30:
hi sir. how about the continuation of this logo? does he like it?

uholeschak, 2018-11-30:
Sorry I forgot about it in the meantime, at the moment I am very busy.
I have asked our designer, basically he thinks it's ok, but the mentioned that color gradients are not modern any more ...

mansya, 2018-12-11:
hi sir, sorry about the long time. I am a little busy with my environment. so where is this color? and you can choose colors as you like between the four colors.
[image]

harryberlin, 2018-12-11:
looks a bit like BOSCH logo.

uholeschak, 2018-12-11:
Perhaps it's best to keep the current logo at the moment, since we both have no real time to change it ...

mansya, 2018-12-15:
if this logo is only found on the file [Resources](https://github.com/uholeschak/ediabaslib/tree/master/BmwDeepObd/Resources) and [Images](https://github.com/uholeschak/ediabaslib/tree/master/docs/Images) I can change it. then I can do pull requests. How ? do you want me to try it?

uholeschak, 2018-12-16:
I think partially changing the logo makes no sense, it should be consistent.
Last time updating all locations was a work of several days, and had the bad the bad effect that,
that people did not find the app in the PlayStore any more.

mansya, 2018-12-22:
if it impacts like that. it would be better to stay consistent with the old logo. Thank you sir

closed 2018-12-22 by mansya