uholeschak/ediabaslib #29: Global settings not saved on Android application close/deactivate

zv2006, 2018-11-08:
Sporadic issue. 
Steps:
1. Open application
2. Open General Settings, Change any settings item (i usually set Keep Screen Bright to keep screen unlocked)
3. After successful connection and communication - disconnect. 
4. Open application after some time (usually after several hours in my case)
5. General settings are reset to its defaults

uholeschak, 2018-11-08:
The settings are store when the App is closed normally. If the app is terminated by another unusual way (killed) the settings will be not stored. Perhaps there is a program in the background that terminates the app after some time.

zv2006, 2018-11-08:
Application is usually not killed or closed manually. I just press phone home button after disconnect
I am not using any special "task management" soft. It is unmodified/unrooted Sony X Compact.

uholeschak, 2018-11-08:
Are you closing the general settings page and return to the main page afterwards?
Settings are stored when closing the main page.
When closing the app while the general settings page is open the settings are not stored.

zv2006, 2018-11-08:
I am closing the general settings and continue to use main page (connect to the car, etc.). Then when i am done, i am pressing phone home button or just power button.

As i said, the problem is very sporadic and it is hard to specify exact steps-to-reproduce. But usually, if i change settings during my morning drive, it will be resetted when i start app for evening drive.

zv2006, 2018-11-08:
Additional info: application itself, configurations and SGBD files are installed on SD card. 
As an idea - could it be permission related issue with just hided access Exception in setting persistence process?

uholeschak, 2018-11-08:
Are the settings completely lost or is only the last change lost?
If they are lost completely (like after app has been reinstalled) there must be another problem.

zv2006, 2018-11-08:
> 
> 
> Are the settings completely lost or is only the last change lost?
> If they are lost completely (like after app has been reinstalled) there must be another problem.

Usually i use only one setting in General Section. So i will perform additional tests.

uholeschak, 2018-11-08:
What I could do is the store the settings directly after closing the general settings page.
Maybe poweroff is not closing the app correctly but switches off immediately.

uholeschak, 2018-11-08:
Maybe I have found the reason:
The settings of the general section are only stored in OnDestroy of the general settings activity, but this callback my be delayed. I will store it now earlier.

zv2006, 2018-11-08:
Great. Will test it in next build

uholeschak, 2018-11-08:
Ok, it's already modified, it will close this for now, reopen the issue when it still persist in the next version.

closed 2018-11-08 by uholeschak

uholeschak, 2018-11-10:
There is a new beta with this fix, perhaps you could try it. Activate with:
https://play.google.com/apps/testing/de.holeschak.bmw_deep_obd

zv2006, 2018-11-12:
Issue still unresolved in 1.2.138

uholeschak, 2018-11-12:
Then please tell me how to reproduce the issue.

zv2006, 2018-11-13:
As i said earlier, the problem is very sporadic. 
If i change settings, exit application and open it again almost immediately (or in 5-10 minutes) - then everything is perfect.
But, if i open it after several hours - with probability 100% settings are changed to defaults.

I will continue to test to localize the issue.

uholeschak, 2018-11-13:
What means default? Is the behavior of the app then similar to a complete reinstall? Are the initial messages displayed again?

zv2006, 2018-11-13:
"default" i mean General section settings only. When i open application in several hours i do not see default messages (like select brand, etc). I only see "compiling user code", the same message when i select new configuration.

zv2006, 2018-11-13:
It could be issues with my platform, of course. BTW, where are settings stored physically?

uholeschak, 2018-11-13:
Seems to be a file:
https://stackoverflow.com/questions/6146106/where-are-shared-preferences-stored

zv2006, 2018-11-13:
/data folder is not avvessible on non-rooted devices. I will investigate it on rooted device later.

uholeschak, 2018-11-13:
It seems that there are some platforms that have problems with the settings:
https://stackoverflow.com/questions/7943573/android-shared-preferences-are-lost-sometimes
Maybe I could use one of these workarounds there ...

zv2006, 2018-11-13:
I was able to pull private data from non-rooted device via backup decryption. 

Attached ZIP with:
good_settings.xml - settings set to correct values (LockComm key), application exit.
lost_settings.xml - after some time application started again, settings are lost in general section, and file reflects shared prefs

[shared_prefs.zip](https://github.com/uholeschak/ediabaslib/files/2575829/shared_prefs.zip)

uholeschak, 2018-11-13:
The app ID has changed in the second file seems that the settings were lost completely in the meantime.

uholeschak, 2018-11-13:
The app ID could be the right hint, it seems that the static part of the app has terminated in the meantime.

zv2006, 2018-11-13:
I have added application to Sony power saving exceptions list to test its termination in background

uholeschak, 2018-11-13:
I have now uploaded a new beta 1.2.139 (will need some time to update), could you check if there is any change?

zv2006, 2018-11-13:
Ok, looks like adding app to power exceptions list did the trick. I will check 1.2.139 build as well. With and without this power related workaround.

zv2006, 2018-11-14:
Looks like 1.2.139 build fix the issue.

uholeschak, 2018-11-14:
Thanks for the info, seems to be the static part of the app in memory.