uholeschak/ediabaslib #34: Mock OBD Adapter

bang-uin, 2019-02-08:
Hi @uholeschak,

thx for your great work! I'm very thankful for it.

I'm struggling a bit with one of my page specifications which shows values of multiple ECUs and hence has multiple Job nodes. My problem is that i want to show some values from engine ECU then some of gear box ECU and then again from engine ECU which leads to the first engine ECU values not showing up. I'm using user defined code to format the values.
Do you by any chance have a mock OBD adapter to be able to test the page without having a real OBD interface connected?

Thx in advance

uholeschak, 2019-02-08:
I am using a car simulation in hardware, that operates with a standard BT adapter.

uholeschak, 2019-02-08:
There is sample file with multiple ECUs:
https://github.com/uholeschak/ediabaslib/blob/master/BmwDeepObd/Xml/E61R/Combined.ccpage

closed 2019-02-08 by uholeschak

bang-uin, 2019-02-08:
What a pity. I looked through the code and it seems not trivial to implement a code based OBD connector.
Anyway, i now adopted my page specification according to the example you gave above and gonna test it later.
Maybe you could have a quick look at it?
Just in case i made something wrong.

```
<?xml version="1.0" encoding="utf-8"?>
<fragment xmlns="http://www.holeschak.de/BmwDeepObd" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://www.holeschak.de/BmwDeepObd BmwDeepObd.xsd">
    <page name="Dashboard" fontsize="medium" display-mode="grid" logfile="DASHBOARD.log">
        <strings>
            <string name="Dashboard">Dashboard</string>
            <string name="LABEL_STAT_MOTORDREHZAHL_MWB_WERT">Enigne speed [1/min]</string>
            <string name="LABEL_STAT_MOTORTEMP_MWB_WERT">Engine temperature [°C]</string>      
            <string name="LABEL_STAT_OEL_TEMP_MWB_WERT">Oil temperature [°C]</string>
            <string name="LABEL_STAT_GETRIEBETEMPERATUR_WERT">Gearbox temp. [°C]</string>
            <string name="LABEL_STAT_KUEHLMITTELTEMP_WERT">Coolant temp. [°C]</string>
            <string name="LABEL_STAT_LAMBDA_INTEGRATOR_1_MWB_WERT">Lambda control factor 1</string>
            <string name="LABEL_STAT_LAMBDA_INTEGRATOR_2_MWB_WERT">Lambda control factor 2</string>     
            <string name="LABEL_STAT_UBATT_MWB_WERT">Voltage [V]</string>
            <string name="LABEL_STAT_GANG">Current gear</string>
        </strings>
            <strings  lang="de">
            <string name="Dashboard">Dashboard</string>
            <string name="LABEL_STAT_MOTORDREHZAHL_MWB_WERT">Drehzahl [U/min]</string>	  
            <string name="LABEL_STAT_MOTORTEMP_MWB_WERT">Motortemperatur [°C]</string>
            <string name="LABEL_STAT_OEL_TEMP_MWB_WERT">Öltemperatur [°C]</string>	
            <string name="LABEL_STAT_GETRIEBETEMPERATUR_WERT">Getriebetemperatur [°C]</string>  
            <string name="LABEL_STAT_KUEHLMITTELTEMP_WERT">Kühlmitteltemperatur [°C]</string>
            <string name="LABEL_STAT_LAMBDA_INTEGRATOR_1_MWB_WERT">Lambdaregelfaktor 1</string>
            <string name="LABEL_STAT_LAMBDA_INTEGRATOR_2_MWB_WERT">Lambdaregelfaktor 2</string>	  
            <string name="LABEL_STAT_UBATT_MWB_WERT">Batteriespannung [V]</string>
            <string name="LABEL_STAT_GANG">Aktueller Gang</string>
        </strings>
        <jobs>
            <job id="1" sgbd="ms430ds0" name="STATUS_MESSWERTEBLOCK" results="STAT_MOTORDREHZAHL_MWB_WERT;STAT_MOTORTEMP_MWB_WERT;STAT_OEL_TEMP_MWB_WERT">
                <display name="LABEL_STAT_MOTORDREHZAHL_MWB_WERT" 
                        result="STAT_MOTORDREHZAHL_MWB_WERT" 
                        log_tag="STAT_MOTORDREHZAHL_MWB_WERT" 
                        grid-type="simple-gauge-round"
                        min-value="0" 
                        max-value="7000" />
                <display name="LABEL_STAT_MOTORTEMP_MWB_WERT" 
                        result="STAT_MOTORTEMP_MWB_WERT" 
                        log_tag="STAT_MOTORTEMP_MWB_WERT" 
                        grid-type="simple-gauge-round" 
                        min-value="-20" 
                        max-value="150" />
                <display name="LABEL_STAT_OEL_TEMP_MWB_WERT" 
                        result="STAT_OEL_TEMP_MWB_WERT" 
                        log_tag="STAT_OEL_TEMP_MWB_WERT" 
                        grid-type="simple-gauge-round" 
                        min-value="-20" 
                        max-value="150" />
            </job>
            <job id="2" sgbd="gs8604" name="STATUS_IO_LESEN" results="STAT_GETRIEBETEMPERATUR_WERT">
                <display name="LABEL_STAT_GETRIEBETEMPERATUR_WERT" 
                        result="STAT_GETRIEBETEMPERATUR_WERT" 
                        log_tag="STAT_GETRIEBETEMPERATUR_WERT" 
                        grid-type="simple-gauge-round" 
                        min-value="-20" 
                        max-value="150" />
            </job>
            <job id="3" sgbd="iki" name="STATUS_CAN_KUEHLMITTELTEMP_LESEN" results="STAT_KUEHLMITTELTEMP_WERT">
                <display name="LABEL_STAT_KUEHLMITTELTEMP_WERT" 
                        result="STAT_KUEHLMITTELTEMP_WERT" 
                        log_tag="STAT_KUEHLMITTELTEMP_WERT"
                        grid-type="simple-gauge-round" 
                        min-value="-20" 
                        max-value="150" />
            </job>
[... 53 lines cut ...]
                        Int64 value = ActivityMain.GetResultInt64(resultDict, resultName, 0, out found);
                        if (found) {
                            result = value.ToString();
                            textColor = ((value < 80 || value > 110) ? Android.Graphics.Color.Red : Android.Graphics.Color.Green);
                        }
                        break;
                    }
                        
                    case "4#STAT_UBATT_MWB_WERT": {
                        double value = ActivityMain.GetResultDouble(resultDict, resultName, 0, out found);
                        if (found) {
                            result = value.ToString("0.0", ActivityMain.Culture);
                            textColor = (value < 11.0 ? Android.Graphics.Color.Red : Android.Graphics.Color.Green);
                        }
                        break;
                    }
                }
                return result;
            }
        }
        ]]>
        </code>
    </page>
</fragment>
```

uholeschak, 2019-02-08:
That's difficult to see at the first glance I expect to have to make several tries ...

bang-uin, 2019-02-08:
Ok, thx. I will try it later.

rs38, 2020-02-02:
is there a special reason why an obd connection is needed to browse through the ECUs, jobs and sub commands? the same is true for the original BMW standard tools btw

uholeschak, 2020-02-02:
Most .grp and .prg files communicate with the ECU during initialization.

uholeschak, 2020-02-02:
At least in the app the EDIABAS tool has an offline mode with restricted functions.