uholeschak/ediabaslib #39: Chipset des Adapters vor dem Kauf rausfinden

EliasKotlyar, 2019-04-14:
Hallo Herr Holeschak

Gibt es eine Möglichkeit vor dem Kauf eines "noName" ELM327 den Chipsatz rauszufinden?  Es geht im wesentlichen dadrum, das ich einen ELM Adapter mit BC-417 erwerben möchte. Pickit3 ist vorhanden und ich weiß wie man solche Devices flasht.
 
Auf den gängigen Seiten fand ich genau einen Händler, der eine "genauere" Beschreibung ink PCB-Bilder hatte. Ich habe mal seine Bilder kopiert - eventuell können sie mir ja sagen welcher Chipsatz was ist. Es gibt bei ihm 2 verschiedene Varianten. Eine mit grünem PCB, und eine mit rotem:

[image]
[image]

[image]

[image]

Ich gehe davon aus, das auf dem roten PCB ein HC05 Chip verbaut ist(dadrauf deutet die Markierung des Chips:m160309). Dieser sollte Pin sowie Software-Kompatibel zu HC04 sein. Wissen die Open-Source Firmware auch damit läuft? 

Vielen Dank im Voraus,
Grüße aus München
Elias

uholeschak, 2019-04-14:
Hallo,
die Roten haben meisten die beste Qualität.

closed 2019-04-14 by uholeschak

nicolai1905, 2020-03-18:
please help me to flash elm327 v1.5 Bluetooth version 
https://github.com/uholeschak/ediabaslib/blob/master/docs/Replace_ELM327_HC04_Firmware.md

i dont understand how to flash . please help me to flash easy way . the elm327 i put on obd port my car after connecting to computer via software flash is possible ?? i need to replace firmware for coding my bmw . please help to replace this firmware easy way

uholeschak, 2020-03-18:
Sorry there is no easy way and it depends from the adapter. This is for experts only

ba114, 2020-05-04:
has anyone found a new adapter that shares the same header pins for programming?
If so, please share the link

uholeschak, 2020-05-04:
The china adapters change every day. If you order another one you may get a different hardware.

shar34, 2020-05-04:
I bought such an adapter (on 26.11.2019), not the same in tutorial, but it works fine.

link: https://aliexpress.ru/item/32893797457.html

[image]
[image]

ba114, 2020-05-04:
Is that the v1.5 product?

shar34, 2020-05-04:
Yes, with PIC18F25K80.

EliasKotlyar, 2020-05-04:
has someone managed to flash the HC05 Bluetooth Module? I mean not the PICKit releted part, but the onboard chip.

As far as i understand, the manufacturing process for this device is:
1. Order a OBD2 Plug with PIC18F25K80 which has some headers  to attach the PIcKIT programmer
2. Order a HC05 Bluetooth Chip 
3. Flash the OBD2 Plug with PicKIT
4. Remove the Bluetooth Chip from the module and replace it with HC05 Chip
5. Replace the firmware of the HC05 with CFW using a TTL-Adapter

I am stuck at point 5. I was doing  the following tutorial:
https://www.instructables.com/id/Upgrade-Your-3-Bluetooth-Module-to-Have-HID-Firmwa/
However, i am stuck with the bluesuite-software, which does not want to recognize the HC05. Does anyone encountered this problem? I am using 220Ohm Resistors.

uholeschak, 2020-05-04:
Basically this is possible, but it's very advanced. I have used a solution with FTDI programmer:
https://github.com/lorf/csr-spi-ftdi/releases
http://www.instructables.com/id/AT-command-mode-of-HC-05-Bluetooth-module/

ba114, 2020-05-04:
> I bought such an adapter (on 11/26/2019), not the same in tutorial, but it works fine.
> 
> link: https://aliexpress.ru/item/32893797457.html
> 
> [image]
> [image]

can you please provide some more photos of the blue wire? also what resistor have you added in to the circuit?

shar34, 2020-05-05:
I have no more photos, the blue wire soldered to the first chip leg (which is the MCLR pin), 512 resistor (http://kiloohm.info/smd3-resistor/512). All wires connected by this scheme, so if your board have no some or all test points, your may solder wires directly to legs:

[image]

ba114, 2020-05-14:
I bought 20 of these... any ideas how to hook them up for programming? they have Bk3432 chip.
Thanks

[image]

shar34, 2020-05-14:
My adapter have BK3432 too and it's working fine with default firmware.

ba114, 2020-05-15:
> My adapter have BK3432 too and it's working fine with default firmware.

You didn't have to flash anything on this adapter?

shar34, 2020-05-15:
> You didn't have to flash anything on this adapter?

Only flash PIC18F25K80 chip.

ba114, 2020-05-15:
> > You didn't have to flash anything on this adapter?
> 
> Only flash PIC18F25K80 chip.

That's my issue, can't flash chip as it's got another board on top and no header pins

denisrohr, 2020-09-06:
> > > You didn't have to flash anything on this adapter?
> > 
> > 
> > Only flash PIC18F25K80 chip.
> 
> That's my issue, can't flash chip as it's got another board on top and no header pins

I unsoldered top pcb, unsoldered the PIC using a heatgun, soldered the pins of the PIC directly to the pickit3, no resistors or capacitors and reprogrammed it using the "default" firmware. Resoldered it to the ELM and I could access the audi via deepobd. Attaching wires to the PIC while its on the pcb does NOT work, since there is some stuff on it that screws with the signal lines.

OrpheeGT, 2021-06-10:
> I unsoldered top pcb, unsoldered the PIC using a heatgun, soldered the pins of the PIC directly to the pickit3, no resistors or capacitors and reprogrammed it using the "default" firmware. [...]

Hi, I have the same board, i unsoldered it :
[image]

I would like to avoid unsolder the chip. 
Using multimeter : 
- Pin1 and Pin20 are wired and have a route to the full top right (currently unsoldered) pin.
- Pin8 (& pin9) is wired to Pin19

Lookiing @shar34 pic, could I do something like this or no way ? :
[image]

Brown links are currently wired pin (tested with multimeter)

uholeschak, 2021-06-10:
The left front pin is the biggest Problem, but I have found to trick:

OrpheeGT, 2021-06-27:
Hello,

I failed to program the module above, but I found the one described in original programming tutorial with RED PCB.

It has a YC1021 bluetooth chip.

I tried first to directly flash YC1021 version of FW, I can't connect to it with Deep OBD.
Then I flashed default FW and it worked.

But I guess I have then default ELM327 FW. I read Deep OBD is able to flash ELM327 FW, and I have the option to do so. But I don't see anywere how to choose YC1021 FW.

I tried to flash ELM327 FW with menu, it worked, and then after reboot Deep ODB suggested to flash a replacement FW for ELM327, so I did it.

It there a way with android to know the current ELM327 FW and see if it is the YC1021 modified one ?

Thanks

uholeschak, 2021-06-27:
If you can't connect the YC1021 baud rate is configured incorrectly. The default FW uses the default baud rate.

OrpheeGT, 2021-06-27:
There is no way to flash YC1021 ELM327 remotely from Deep ODB to enable 115200 speed ? I must solder on it as explained here ?

https://github.com/uholeschak/ediabaslib/blob/master/docs/Replace_Elm327_BT_Mini_Firmware.md

uholeschak, 2021-06-27:
Correct.

drbluetongue, 2021-08-03:
[image]

[image]

Hello!

Has anyone flashed one of these before? BK3231.

drbluetongue, 2021-08-10:
So I found another adaptor that has pins on it. 

Flashed successfully the default file, but no Bluetooth afterwards :/

Avoid this BT chip
[image]

tdashmike, 2021-11-11:
@shar34 

Which firmware did you flash? It looks like the bluetooth chip on that one is a BP09497 but I cannot find any info on it. Thanks.

shar34, 2021-11-11:
> @shar34
> 
> Which firmware did you flash? It looks like the bluetooth chip on that one is a BP09497 but I cannot find any info on it. Thanks.

Hello,

I flashed this firmware: **/CanAdapterElm/default/CanAdapterElm.X.production.unified.hex**, but I have BK3432 chip, not BP09497.

tdashmike, 2021-11-12:
@shar34 Thanks, I tried that and it worked one time and it stopped working. I think I damaged the pads when I was trying to remove the board to see waht bluetooth moduel it was. I'll get a few more and possibly sell the extras on eBay.

neuronet77, 2021-12-30:
I found this adapter in my "cable" collection. Do you know whether this one can be used? I can't see any "HC" chip on it :(

[image]

uholeschak, 2021-12-30:
The Bluetooth chip seems to be somewhere else.

neuronet77, 2021-12-30:
Sorry. had to pray it out to get to the back... :(
[image]

neuronet77, 2021-12-30:
I think I just got a "bingo" :D Now just waiting for the PICkit 3 programmer to be delivered... I saw a picture of you having connected 5 wires on the soldering dots on the one side of the board. It looks really like mine, so the wire colors you show there are the ones from the PICkit 3 cable?

Thanks so much for all you've done here! 

[image]

uholeschak, 2021-12-30:
The photo was not from me, but at least the pin order is identical to PicKit