uholeschak/ediabaslib #40: Job STATUS_MESSWERTBLOCK_0 sometimes does not work

6i6i, 2019-04-20:
Hello,
I got EdiabasTest running on mono on an raspberry pi. I sometimes get the problem, that I cannot execute the job STATUS_MESSWERTBLOCK_0. Other Jobs are running f.ex. STATUS_DIGITAL_0. I am using a maxDiag1 USB adapter for my BMW Z4 E85 with MS45 ECU. I get an ERROR_ECU_CONDITIONS_NOT_CORRECT_OR_REQUEST_SEQUENCE_ERROR
for job status.
See here my commands I executed.
osmc@BMW:~/ediabaslib/EdiabasLib/EdiabasTest/bin/Debug$ sudo mono EdiabasTest.exe --alltypes -s ms450ds0.prg -j STATUS_MESSWERTBLOCK_0 -p /dev/ttyUSB1
API Version: 7.3.0
JOB: STATUS_MESSWERTBLOCK_0
Variant: MS450DS0
DATASET: 0
IGNITIONCURRENT: I: -1 0xFFFF ALL:  -1 255 -1 65535 -1 4294967295 -1
IGNITIONHISTORY: I: -1 0xFFFF ALL:  -1 255 -1 65535 -1 4294967295 -1
JOBNAME: STATUS_MESSWERTBLOCK_0
JOBSTATUS: 
OBJECT: ms450ds0
SAETZE: W: 1 0x0001 ALL:  1 1 1 1 1 1 1
UBATTCURRENT: I: -1 0xFFFF ALL:  -1 255 -1 65535 -1 4294967295 -1
UBATTHISTORY: I: -1 0xFFFF ALL:  -1 255 -1 65535 -1 4294967295 -1
VARIANTE: MS450DS0
DATASET: 1
_TEL_ANTWORT: B8 F1 12 03 7F 21 22 24 
_TEL_AUFTRAG: B8 12 F1 02 21 F0 
JOB_STATUS: ERROR_ECU_CONDITIONS_NOT_CORRECT_OR_REQUEST_SEQUENCE_ERROR

osmc@BMW:~/ediabaslib/EdiabasLib/EdiabasTest/bin/Debug$ sudo mono EdiabasTest.exe --alltypes -s ms450ds0.prg -j STATUS_DIGITAL_0 -p /dev/ttyUSB1
API Version: 7.3.0
JOB: STATUS_DIGITAL_0
Variant: MS450DS0
DATASET: 0
IGNITIONCURRENT: I: -1 0xFFFF ALL:  -1 255 -1 65535 -1 4294967295 -1
IGNITIONHISTORY: I: -1 0xFFFF ALL:  -1 255 -1 65535 -1 4294967295 -1
JOBNAME: STATUS_DIGITAL_0
JOBSTATUS: 
OBJECT: ms450ds0
SAETZE: W: 1 0x0001 ALL:  1 1 1 1 1 1 1
UBATTCURRENT: I: -1 0xFFFF ALL:  -1 255 -1 65535 -1 4294967295 -1
UBATTHISTORY: I: -1 0xFFFF ALL:  -1 255 -1 65535 -1 4294967295 -1
VARIANTE: MS450DS0
DATASET: 1
_TEL_ANTWORT: 62 40 07 00 00 08 42 1B F5 
_TEL_AUFTRAG: B8 12 F1 03 22 40 07 
JOB_STATUS: OKAY
STAT_LL_EIN: I: 0 0x0000 ALL:  0 0 0 0 0 0 0
STAT_SBBHK: I: 0 0x0000 ALL:  0 0 0 0 0 0 0
STAT_SBBHK2: I: 0 0x0000 ALL:  0 0 0 0 0 0 0
STAT_SBBVK: I: 0 0x0000 ALL:  0 0 0 0 0 0 0
STAT_SBBVK2: I: 0 0x0000 ALL:  0 0 0 0 0 0 0

Do you have any idea why this happens?
With the same setup it is sometimes possible to read STATUS_MESSWERTBLOCK_0 and sometimes not.
Is there any job I have to run before it is possible to run STATUS_MESSWERTBLOCK_0?

I can connect using INPA without any problems with the same interface.

Thank you BR
Martin

6i6i, 2019-04-20:
I just checked it with DeepOBD with the same hardware on my Android Phone and get the exact same error:
[image]
[image]

6i6i, 2019-04-20:
One additional thing I just found out :-) is that if I connect with INPA and execute STATUS_MESSWERTBLOCK_0 i get the data immediately. If I disconnect afterwards and do not switch off the ignition and connect my pi again I can read the data without problems:
osmc@BMW:~/ediabaslib/EdiabasLib/EdiabasTest/bin/Debug$ sudo mono EdiabasTest.exe --alltypes -s ms450ds0.prg -j STATUS_MESSWERTBLOCK_0 -p /dev/ttyUSB1
API Version: 7.3.0
JOB: STATUS_MESSWERTBLOCK_0
Variant: MS450DS0
DATASET: 0
IGNITIONCURRENT: I: -1 0xFFFF ALL:  -1 255 -1 65535 -1 4294967295 -1
IGNITIONHISTORY: I: -1 0xFFFF ALL:  -1 255 -1 65535 -1 4294967295 -1
JOBNAME: STATUS_MESSWERTBLOCK_0
JOBSTATUS: 
OBJECT: ms450ds0
SAETZE: W: 1 0x0001 ALL:  1 1 1 1 1 1 1
UBATTCURRENT: I: -1 0xFFFF ALL:  -1 255 -1 65535 -1 4294967295 -1
UBATTHISTORY: I: -1 0xFFFF ALL:  -1 255 -1 65535 -1 4294967295 -1
VARIANTE: MS450DS0
DATASET: 1
_TEL_ANTWORT: 61 F0 7E 00 00 00 00 00 04 19 51 00 52 00 65 00 35 00 00 00 2E 80 03 02 00 00 27 80 D9 40 
_TEL_AUFTRAG: B8 12 F1 02 21 F0 
JOB_STATUS: OKAY
STAT_MESSWERT0_EINH: V
STAT_MESSWERT0_TEXT: Spannung Kl.87
STAT_MESSWERT0_WERT: R: 12.796890624 ALL:  12 12 12 12 12 12 12.796890624
STAT_MESSWERT1_EINH: km/h
STAT_MESSWERT1_TEXT: Geschwindigkeit
STAT_MESSWERT1_WERT: R: 0 ALL:  0 0 0 0 0 0 0
STAT_MESSWERT2_EINH: 1/min
STAT_MESSWERT2_TEXT: Motordrehzahl
STAT_MESSWERT2_WERT: R: 0 ALL:  0 0 0 0 0 0 0
STAT_MESSWERT3_EINH: 1/min
STAT_MESSWERT3_TEXT: LLR – Solldrehzahl
STAT_MESSWERT3_WERT: R: 1049 ALL:  25 25 1049 1049 1049 1049 1049
STAT_MESSWERT4_EINH: C
STAT_MESSWERT4_TEXT: Motortemperatur
STAT_MESSWERT4_WERT: R: 12.750010368 ALL:  12 12 12 12 12 12 12.750010368
STAT_MESSWERT5_EINH: C
STAT_MESSWERT5_TEXT: Kühlmitteltemperatur Kühlerausgang
STAT_MESSWERT5_WERT: R: 13.500010496 ALL:  13 13 13 13 13 13 13.500010496
STAT_MESSWERT6_EINH: C
STAT_MESSWERT6_TEXT: Ansauglufttemperatur
STAT_MESSWERT6_WERT: R: 27.750012928 ALL:  27 27 27 27 27 27 27.750012928
STAT_MESSWERT7_EINH: C
STAT_MESSWERT7_TEXT: (Motor) - Öltemperatur
STAT_MESSWERT7_WERT: R: 13 ALL:  13 13 13 13 13 13 13
STAT_MESSWERT8_EINH: kg/h
STAT_MESSWERT8_TEXT: Luftmasse
STAT_MESSWERT8_WERT: R: 0 ALL:  0 0 0 0 0 0 0
STAT_MESSWERT9_EINH: hPa
STAT_MESSWERT9_TEXT: Umgebungsdruck
STAT_MESSWERT9_WERT: R: 987.03515904 ALL:  -37 219 987 987 987 987 987.03515904
STAT_MESSWERTA_EINH: °DK
STAT_MESSWERTA_TEXT: Drosselklappe Sollwert
STAT_MESSWERTA_WERT: R: 5.61614977 ALL:  5 5 5 5 5 5 5.61614977
STAT_MESSWERTB_EINH: %
STAT_MESSWERTB_TEXT: Fahrpedalwert
STAT_MESSWERTB_WERT: R: 0 ALL:  0 0 0 0 0 0 0
STAT_MESSWERTC_EINH: V
STAT_MESSWERTC_TEXT: Spannung Drosselklappe Poti1
STAT_MESSWERTC_WERT: R: 0.77144448 ALL:  0 0 0 0 0 0 0.77144448
STAT_MESSWERTD_EINH: V
STAT_MESSWERTD_TEXT: Spannung Drosselklappe Poti2
STAT_MESSWERTD_WERT: R: 4.24294464 ALL:  4 4 4 4 4 4 4.24294464

uholeschak, 2019-04-20:
Please attach the trace files for the case with failures.

uholeschak, 2019-04-20:
If this is no communication problem, please check the INPA traces to see which jobs they are using.

6i6i, 2019-04-20:
Thank you very much for the quick response. Can you provide me a quick introduction how I can get the INPA trace files?

uholeschak, 2019-04-20:
At the moment I don't remember how to enable them, one solution would be to replace the api32.dll with the EdiabasLib version and enable logging there or simply log the telegrams on the line.

6i6i, 2019-04-20:
Ok. I'll give it a try afterwards. Thank you very much for the support!

uholeschak, 2019-04-20:
You could try to set the corresponding flags in ediabas.ini, but I think INPA overrides them afterwards.

uholeschak, 2019-04-20:
If I remember correctly, I have used the following flags in EdiabsLib.config to force tracing:
```xml
    <add key="ApiTrace" value="1" />
    <add key="IfhTrace" value="2" />
    <add key="LockTrace" value="1" />
```
You have to replace the api32.dll of course.
If INPA has the same problem with the replaced dll, it's probably a bug in EdiabsLib.

6i6i, 2019-04-20:
Hi, I have no compiled API32.dll so far. But I managed to enable trace in INPA by setting ApiTrace         = 8
IfhTrace         = 3 in file EDIABAS.INI.

I attached the trace files from INPA and from EdiabasTest Commandline tool before and after the execution of INPA. For me the trace files are looking quite different, but maybe you can read something out of it?
What I can definitely say is that if I execute INPA and  execute the reading of STATUS_MESSWERTBLOCK_0 it will afterwards work with EdiabasTest too if I do not put the ignition off in the meanwhile. It seems that somehow INPA is executing something additional to put the ECU in a "correct mode".

Thank you
[ediabastrace.zip](https://github.com/uholeschak/ediabaslib/files/3100612/ediabastrace.zip)

uholeschak, 2019-04-20:
Thanks, I will check that when I have some time, but it seems that the important part is missing in the INPA trace (part been the api.trc and apiold.trc)

uholeschak, 2019-04-20:
You have get the job before the first:
```
[23:16:58]	apiJob(
			"MS450DS0",
			"STATUS_MESSWERTBLOCK_0",
			"",
			""
			)
```

6i6i, 2019-04-20:
Ok. I understand seems that INPA somehow has overwritten a part of the trace file. :-(. I'll check this tomorrow.
Thank you

6i6i, 2019-04-21:
Hi,
I found out, that INPA first executes MESSWERTBLOCK_0_SCHREIBEN before it executes STATUS_MESSWERTBLOCK_0. So I tried this and it seems to work now.
Thank you very much for guiding me to this solution and this great piece of software from your side! :-)

BR

uholeschak, 2019-04-21:
Thanks for the info, does this job require any parameters?
Perhaps I could add this to the documentation section.

closed 2019-04-21 by uholeschak

6i6i, 2019-04-21:
No it does not require any parameters. But I think the same is necessary for the other STATUS_MESSWERTBLOCK_X.

uholeschak, 2019-04-21:
Thanks, I will check if there is also a write command, perhaps I could add it to the configuration generator of the app.

uholeschak, 2019-04-21:
It seems to to be a special thing only for MS450DS0, there MESSWERTBLOCK_0_SCHREIBEN to MESSWERTBLOCK_7_SCHREIBEN is existing.

6i6i, 2019-04-22:
Ok. Strange. But they would have some reason for this propably. :-D

uholeschak, 2019-04-22:
I have now added some info to the documentation section, at least this knowledge will not be lost ...

6i6i, 2019-04-22:
OK that's good. Thank you very much again for this awesome software and the quick and kind support!