uholeschak/ediabaslib #41: Ediabaslib WLAN and ISTA+

Vernerio, 2019-04-22:
I have used it with large variety of cars.
Mainly it works fine and very easy to connect (compared to ENET). But sometimes it lost connection after succesful identification to some modules (example DDE). Also Fault memory  delete not work, because new ISTA+ want's to control ignition voltage terminal change during memory delete. 99% times terminal change fail with wlan adapter and also memory delete fail then.
With ICOM it works every time, ISTA turn car off and back ON and continue delete.
Sorry I dont have any trace logs now, but I can arrange on some day.

uholeschak, 2019-04-22:
Without any traces it will get difficult ...

uholeschak, 2019-04-22:
Do you have the same problem when using a cable? If not then probably only your WiFi connection is disturbed.

uholeschak, 2019-04-25:
Probably a WiFi connection problem, no action required.

closed 2019-04-25 by uholeschak

Vernerio, 2019-05-15:
IFH -Trace
`executeJob('d_cas.grp'): 'IDENTIFIKATION'
executeJob('d_cas.grp'): 'INITIALISIERUNG'
Connect
Sending: 'VirtualBox Host-Only Network': Ip=192.168.56.1 Mask=255.255.255.0 Broadcast=192.168.56.255
Sending: 'VMware Network Adapter VMnet1': Ip=192.168.84.1 Mask=255.255.255.0 Broadcast=192.168.84.255
Sending: 'VMware Network Adapter VMnet8': Ip=192.168.255.1 Mask=255.255.255.0 Broadcast=192.168.255.255
Sending: 'WLAN': Ip=192.168.100.246 Mask=255.255.255.0 Broadcast=192.168.100.255
Sending: 'Loopback Pseudo-Interface 1': Ip=127.0.0.1 Mask=255.0.0.0 Broadcast=127.255.255.255
Received: IP=192.168.100.195
Connected
 (ENET CommParameter Host=auto:all, Tester=0xF4, ControlPort=6811, DiagPort=6801): 0000010F 0001C200 000004B0 00000014 0000000A 00000002 00001388 
Answer length: 0000 0000
 (Send): 82 40 F1 1A 80 
 (Resp): 83 F1 40 7F 1A 11 5E 
executeIdentJob failed
ResolveSgbdFile: No variant found
executeJob('g_zgw.grp'): 'IDENTIFIKATION'
executeJob('g_zgw.grp'): 'INITIALISIERUNG'
 (ENET CommParameter Host=auto:all, Tester=0xF4, ControlPort=6811, DiagPort=6801): 00000110 0007A120 000003E8 00000000 00000000 000003E8 000003E8 000004B0 00000000 00001388 00000002 000003E8 00000000 0000000A 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000001 
Answer length: 0000 0000
 (Send): 83 10 F1 22 F1 50 
 (Resp): 86 F1 10 62 F1 50 0F 13 10 5C 
executeJob('zgw_01.prg'): 'STATUS_VIN_LESEN'
executeJob('zgw_01.prg'): 'INITIALISIERUNG'
 (ENET CommParameter Host=auto:all, Tester=0xF4, ControlPort=6811, DiagPort=6801): 00000110 0007A120 000003E8 00000000 00000000 000003E8 000003E8 000004B0 00000000 00001388 00000002 000003E8 00000000 0000000A 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000001 
Answer length: 0000 0000
 (Send): 83 10 F1 22 F1 90 
 (Resp): 94 F1 10 62 F1 90 57 42 41 35 4C 35 31 30 38 47 47 30 38 39 39 39 39 7B 
executeJob('zgw_01.prg'): 'STATUS_VCM_GET_FA'
 (Send): 83 10 F1 22 3F 06 
 (Resp): 80 F1 10 A6 62 3F 06 00 8B 03 41 74 55 99 04 51 56 C5 51 42 15 52 B2 3B F8 84 63 85 18 E2 46 38 D1 8E 44 AB 45 2C 21 4B 68 92 41 54 92 4D 24 D0 49 46 12 55 54 E1 9D 38 70 4E 1D 13 45 65 21 D1 4B A2 53 54 94 49 85 13 45 46 11 51 86 14 65 45 19 59 58 51 55 04 95 41 85 53 59 55 18 5A EC D6 D6 85 B7 89 64 19 59 75 96 65 85 D6 45 88 78 62 16 58 AE 16 33 B5 8D 26 63 49 D8 D2 C6 14 49 85 50 66 18 59 65 35 AE C8 48 51 41 5B 25 92 53 2F B7 38 40 00 00 14 D9 E6 EF 80 D7 98 98 BB 38 BF DF 10 0D 5D B8 0D 9F A3 5B 25 E4 
executeJob('g_cas.grp'): 'IDENTIFIKATION'
executeJob('g_cas.grp'): 'INITIALISIERUNG'
 (ENET CommParameter Host=auto:all, Tester=0xF4, ControlPort=6811, DiagPort=6801): 00000110 0007A120 000003E8 00000000 00000000 000003E8 000003E8 000004B0 00000000 00001388 00000002 000003E8 00000000 0000000A 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000001 
Answer length: 0000 0000
 (Send): 83 40 F1 22 F1 50 
 (Resp): 86 F1 40 62 F1 50 0F 10 01 7A 
executeJob('cas4_2.prg'): 'IDENT'
ReadAllJobs file not found: C:\Program Files\Rheingold\Ecu\_cas4.prg
executeJob('cas4_2.prg'): 'INITIALISIERUNG'
 (ENET CommParameter Host=auto:all, Tester=0xF4, ControlPort=6811, DiagPort=6801): 00000110 0007A120 000003E8 00000000 00000000 000003E8 000003E8 000004B0 00000000 00001388 00000002 000003E8 00000000 0000000A 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000001 
Answer length: 0000 0000
 (Send): 83 40 F1 22 F1 50 
 (Resp): 86 F1 40 62 F1 50 0F 10 01 7A 
 (ArgBinary): 49 44 3B 30 78 41 43 35 31 3B 53 54 52 3B 33 
executeJob('cas4_2.prg'): 'STEUERN_ROUTINE'
 (Send): 85 40 F1 31 01 AC 51 03 
 (Resp): 84 F1 40 71 01 AC 51 24 
SetError: EDIABAS_BIP_0010
executeJob('zgw_01.prg'): 'STATUS_VCM_I_STUFE_LESEN'
executeJob('zgw_01.prg'): 'INITIALISIERUNG'
 (ENET CommParameter Host=auto:all, Tester=0xF4, ControlPort=6811, DiagPort=6801): 00000110 0007A120 000003E8 00000000 00000000 000003E8 000003E8 000004B0 00000000 00001388 00000002 000003E8 00000000 0000000A 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000001 
Answer length: 0000 0000
 (Send): 83 10 F1 22 10 0B 
 (Resp): B1 F1 10 62 10 0B 46 30 31 30 10 0B 01 F7 46 30 31 30 0F 0B 01 F7 46 30 31 30 0F 07 01 F8 00 14 65 A2 CD A4 99 C1 D8 12 52 F8 91 79 9D 14 54 49 3F 5F EC 06 EA 
executeJob('zgw_01.prg'): 'STATUS_VCM_GET_ECU_LIST_ALL'
 (Send): 83 10 F1 22 3F 07 
 (Resp): A5 F1 10 62 3F 07 00 20 00 01 02 06 0B 0D 10 12 17 18 19 1C 20 29 2A 30 35 38 39 40 5E 60 63 64 67 72 78 79 A5 A6 A7 A8 87 
executeJob('zgw_01.prg'): 'STATUS_VCM_GET_ECU_LIST_MOST'
 (Send): 83 10 F1 22 3F 11 
 (Resp): 87 F1 10 62 3F 11 00 02 60 63 FF 
executeJob('zgw_01.prg'): 'STATUS_VCM_GET_ECU_LIST_BODY_CAN'
 (Send): 83 10 F1 22 3F 0F 
 (Resp): 8E F1 10 62 3F 0F 00 09 00 0D 10 35 40 63 64 67 72 7A 
executeJob('zgw_01.prg'): 'STATUS_VCM_GET_ECU_LIST_K_CAN'
 (Send): 83 10 F1 22 3F 0E 
 (Resp): 8B F1 10 62 3F 0E 00 06 06 10 20 38 78 79 A0 
executeJob('zgw_01.prg'): 'STATUS_VCM_GET_ECU_LIST_FLEXRAY'
 (Send): 83 10 F1 22 3F 13 
[... 1009 lines cut ...]
 (Resp): 82 F1 63 50 01 27 
NR78(63) removed
 (Send): 82 63 F1 10 03 
 (Resp): 83 F1 63 7F 10 78 DE 
NR78(63) added
[... cut ...] 3 EB 85 58 43 D9 08 AF 60 7C E2 D2 D4 36 8B 4D 12 38 5F 86 9B 0A 08 04 07 1A E8 5A BE E0 A3 78 D5 B6 C9 BC EF E9 FB 42 20 A1 29 91 97 09 A7 86 EA BF BC 4E A2 8D B9 
 (Send): 82 63 F1 10 01 
 (Resp): 82 F1 63 50 01 27 
 (ArgBinary): 44 45 46 41 55 4C 54 
executeJob('swtuds.prg'): 'DIAGNOSE_MODE'
 (Send): 82 63 F1 10 03 
 (Resp): 83 F1 63 7F 10 78 DE 
NR78(63) added
 (Resp): 86 F1 63 50 03 01 2B 01 F4 4E 
NR78(63) removed
 (Send): 82 63 F1 10 01 
 (Resp): 82 F1 63 50 01 27 
Disconnect
Disconnect
`

uholeschak, 2019-05-15:
This seems to be a timeout issue, probably the data is delay over WiFi.
Could you try to change the timeout value in EdibasLib.config (value in ms)?
```xml
        <add key="ObdAddRecTimeout" value="1000"/>
```

Vernerio, 2019-05-15:
Thanks, yes I will report when confirmed.

uholeschak, 2019-05-15:
Have you also tested with a cable? I would expect that there is no problem in this case.