uholeschak/ediabaslib #53: Logging in deepobd is unreliable

djwlindenaar, 2019-10-06:
First of all, thanks for a great app, which is very helpful in finding problems with my car.

I've been taking a lot of logs lately (tracking down an issue) and I have had several occasions where the logs did not get written. 
I found sometimes the checkmark next to logging gets reset and no logging gets written. Also always writing the data to the same file can be problematic for me.

I'd propose to create an option to always log the data (never automatically reset this option) and to create a new logfile with timestamp in the filename each time the app connects to the dongle.

Also (unrelated), in case a job result contains newlines, the CSV file is not (easily) readable anymore. I'd propose to replace any newline characters in job results with ; or something similar.

Best regards
Daniel

uholeschak, 2019-10-06:
For storing the logging settings there is an option the global settings.
When using the symbol {D} in the log file name it with be replaced with the current date and time.
https://github.com/uholeschak/ediabaslib/blob/master/docs/Page_specification.md

closed 2019-10-06 by uholeschak

djwlindenaar, 2019-10-11:
I missed the {D} option.

However, the main issue is that the app sometimes decides on its own to disable logging. Which is unreliable. Sometimes it remembers the enabled log setting across restarts of the app and sometimes not. 

I want to have a reliable way to keep logging enabled without having to manually check it each time.

Best regards
Daniel

uholeschak, 2019-10-11:
Have you enabled storing the log settings?
If yes and it still looses the settings please specify how to reproduce the issue.

djwlindenaar, 2019-10-11:
No, never knew that setting existed... Let me check again.

Thanks!

uholeschak, 2019-10-11:
Without this setting the checkbox is cleared at every app start to prevent SD card memory overflow.
This is no bug.

djwlindenaar, 2019-10-11:
I understand that now. I completely missed that setting. Maybe a hint in the logging menu would be helpful, though.

Thanks