uholeschak/ediabaslib #57: Problems E61 LCI

Timmmeyyy, 2019-12-18:
Hi I have some issues to read the ecus, i have an vgate icar2 tool. I tried to read the ecus on an bmw e61 lci build in late 2007.
 I take an screenshot, i pretty sure that my car dont have k-line, because its an lci. 
With the vgate tool and "bimmercode" its works quit fine also with the apps "torque" or "car scanner".  

With an other tool i have i can work without any problems with inpa and tool32 via usb and my notebook. 

Maybe, you have any ideas where my mistake is. 

Greetz Tim 
 
[image]

uholeschak, 2019-12-18:
The screenshot does not help, send a trace file and publish your app id.

Timmmeyyy, 2019-12-19:
Good Morning, how can i get the tracefile? Greetz Tim

mark-sch, 2019-12-19:
With a little passion everyone is able to figure out. Ask valueable questions to the valueable time of Ulrich ;-) He is commiting awesome investigations to github atm.

Timmmeyyy, 2019-12-19:
Figure it out, sorry for that stupid question :D. I will create it later an give you an feedback with that informations. Greetz Tim

uholeschak, 2019-12-19:
You could simply send it with the app.

Timmmeyyy, 2019-12-19:
Hi, i send you an email with the tracefile. The Screenshot below show the ID.

Greetz Tim 

[image]

uholeschak, 2019-12-19:
What means email? If you send it with the app, it's stored in a database but there is nothing.

uholeschak, 2019-12-19:
I the found the data now. Basically the detection is working in the first step, VIN and construction date is extracted correctly. But when trying to find the ECUs with a broadcast there is no response.

Timmmeyyy, 2019-12-19:
Ah sorry no email my mistake :D. You have any idea what i can try else? Its is not the k-line problem, its dcan?

greetz

uholeschak, 2019-12-19:
Perhaps you adapter has problems with the broadcast responses. There will be a flood of answers, maybe it's too much. Perhaps you should try with an USB FTDI adapter, this is more stable.

Timmmeyyy, 2019-12-19:
My Dcan usb cable works fine with inpa. You have an alternativ wifi adapter which should work fine with your app an my car?

uholeschak, 2019-12-19:
Not at the moment, but try the USB cable with the app. INPA not using broadcasts, is it working also with ISTA-D? Maybe the ECU for the broadcasts is damaged.

Timmmeyyy, 2019-12-19:
I give you a feedback tomorrow . Greetz

Timmmeyyy, 2019-12-20:
Hi with the usb cable it works pretty fine, so it is a Problem with the vgate tool? Before i bought it i see a lot of YouTube vids where people use this tool? I send you the tracefile today withe cable use.  Maybe you have an solution how can i use the vgate tool? Greetz Tim

uholeschak, 2019-12-20:
Thanks for the info, but I see no solution for your adapter, you could only try to buy different ones.

closed 2019-12-20 by uholeschak