uholeschak/ediabaslib #60: Tablet doesn't get a connection

rs38, 2020-02-26:
pretty old  and slow GT P 5110
https://en.wikipedia.org/wiki/Samsung_Galaxy_Tab_2_10.1
latest Deep OBD
connected with gender changer to FTDI USB cable 
Android 5.1.1 custom (https://slimroms.org/) witch is probably rooted and has extended debug settings.

Deep OBD starts and extracts fine, but whether I use the config generator or ediabas tool and choose USB/FTDI it does simply nothing. No warning / error. I am also unable to create trace files or similar. So hard to provide better info for this...

How could I check whether it can talk to the FTDI at all?

uholeschak, 2020-02-26:
Do you get a popup regarding USB connection usage from the app? If not the tablet is probably not usable.

uholeschak, 2020-02-26:
Please test the cable with an up to date Android device. If this is working correctly, your tablet is simply to old and you have to replace it or use a different Android.

closed 2020-02-26 by uholeschak

rs38, 2020-02-27:
you mean a pop up at app startup like "do you want to use the USB with this app"?
No there isn't. Wouldn't it be meaningful to have a error message, if DeapOBD cannot find or reach an USB device?

uholeschak, 2020-02-27:
Yes if there is no popup USB is not available. I think this info is enough.

Crabolog, 2020-03-02:
Am I correct that initial flashing via pickit and powering by 16&4 pins
should make the adapter visible as a bluetooth device without inserting it
into obd?

uholeschak, 2020-03-02:
Bluetooth is independent of the PIC, must be visible always.

Crabolog, 2020-03-02:
Got an adapter with BK3231 and bluetooth visible only while inserted into
obd

uholeschak, 2020-03-02:
Without power supply it is of course not visible.

Crabolog, 2020-03-02:
Nah, powered at home with 12v, not visible

Crabolog, 2020-03-02:
However its reading and flashing

пн, 2 мар. 2020 г. в 18:21, Oleg Crabolog <crabeg23@gmail.com>:

uholeschak, 2020-03-02:
Sorry I can't help you there, your hardware is unknown to me.

rs38, 2020-03-02:
> pretty old and slow GT P 5110
> https://en.wikipedia.org/wiki/Samsung_Galaxy_Tab_2_10.1
> latest Deep OBD
> connected with gender changer to FTDI USB cable
> Deep OBD starts and extracts fine, but whether I use the config generator or ediabas tool and choose USB/FTDI it does simply nothing. No warning / error. I am also unable to create trace files or similar. So hard to provide better info for this...

wanted to add that the reason was the gender changer is not triggering the USB cable to be the slave. So when connecting a true OTG Cable, it works and the pop up is displayed.
Nevertheless it would be kind of helpful to have a check whether the USB connection is working at all or not.

uholeschak, 2020-03-02:
If there is no USB popup the adapter is not detected. Use a newer tablet.

rs38, 2020-03-02:
obviously you didn't get my message: tablet is fine, but attaching a USB OBD2 cable with a gender changer is not working. OTG is required and works fine. OTG defines / reverses master and slave in the USB connection. USB storage was also not possible. may be a beginner's mistake.

uholeschak, 2020-03-02:
Sorry I don't not understand it. It is working now?

rs38, 2020-03-02:
liegt's an der Sprache??? ja, es läuft jetzt mit einem OTG Kabel. nur mit einem weiblich-weiblich Adapter weiß das Tablet nicht, dass es nun mit einem FTDI als slave reden soll. Das war ne Fehlbedienung, die offenbar nicht jedem klar ist.

uholeschak, 2020-03-02:
Prima, dann ist ja alles ok, das kann schon sein, dass es an der Sprache liegt.
Ja dafür sind die OTG Kabel ja auch da.