uholeschak/ediabaslib #61: Slow reading?

doharvey, 2020-03-28:
I have selected all jobs in the ecu and it worked but it takes ~5 sec to refresh, is it normal?

uholeschak, 2020-03-28:
Generally this is not possible to answer, this depends from the adapter type, connection, protocol and vehicle.
ELM327 for example is a very bad choice.

closed 2020-03-28 by uholeschak

GyurkanM, 2021-11-21:
Reading becomes very slow when I enable BMW database, from data updates per second it becomes updates every minute, can you provide any insights on this? When I use manual jobs selected from the me9k_ng4 it is very fast, updates per second. I have tested with 2 different adapters flashed with yc1021 firmware, no change from the default reading speed.

uholeschak, 2021-11-21:
If your phone is slow the data evaluation is time consuming of course with the database. Use faster hardware.

GyurkanM, 2021-11-21:
Hi , the phone is lenovo p2, Snapdragon 625(8cores 2ghz), 4gb ram, so  we  should consider that this is below acceptable performance hardware for DeepOBD usage. Ram usage by deepobd is at maximum 350mb, so you mean likely that the cpu is the bottleneck here? Thanks

uholeschak, 2021-11-21:
Yes the CPU is the problem. If you only use jobs that are not linked to the database it will be faster.