uholeschak/ediabaslib #63: Sorry for notification.

AraHaan, 2020-04-07:
I was made aware that a bug in how I was arraying the bytes would cause the bytes to be output in correct decompressed / compressed length but result in every byte potentially being null since zlib.managed v1.1.2. I plan to release a v1.1.4 to attempt to fix this issue.

Sorry for any potential inconvenience this has caused.

uholeschak, 2020-04-07:
Thanks for the information, I will check this.

closed 2020-04-07 by uholeschak

AraHaan, 2020-04-07:
Alright I did some internal testing using projects that use the fixed build to make an xmplay plugin that handles archive formats that requires zlib to decompress the entries and they seem to decompress fine now. Going to push the changes to the github in a few.

AraHaan, 2020-04-09:
The new release was published earlier.