uholeschak/ediabaslib #65: Execution errors when running certain jobs with Ediabaslib.dll & Ediabaslib Api32.dll

Martyn-Dev, 2020-04-27:
Hi,

I have been using Ediabaslib.dll successfully in a number of my projects, however I believe I may have stumbled upon a bug. I can successfully execute the job in question using identical parameters using Tool32 and the stock api32.dll, however when I execute the job with the ediabaslib api32.dll I get an error. The same error occurs if I repeat the test using EdibasTest.exe too.

I am running the job "FLASH_LOESCHEN" from the SMG2.prg file and passing the parameter "010100000000000000FF0000006000600010200300FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF03".

The errors I receive are "DATATRANSMISSION TO INTERFACE DISTURBED" or "ERROR_ECU_PARAMETER".

Attached are trace logs from the working api32.dll and tool32, along with the traces from the non working ediabaslib api32.dll.
[ediabaslib_logs.zip](https://github.com/uholeschak/ediabaslib/files/4538792/ediabaslib_logs.zip)

Many thanks,

Martyn

uholeschak, 2020-04-27:
Thanks for the info, do I need the ECU for testing?

Martyn-Dev, 2020-04-27:
Unfortunately yes. 'FLASH_LOESCHEN' is a destructive process as it will erase the flash, however I have a bench setup for testing. I can arrange a remote session or something perhaps?

uholeschak, 2020-04-27:
Perhaps if I have the ECU response telegrams I could try to simulate it.

Martyn-Dev, 2020-04-27:
I can set the logging to the highest and execute the working process in WinKFP if you wish?

uholeschak, 2020-04-27:
Thanks at least you could try

Martyn-Dev, 2020-04-27:
API trace level 8 / IFH trace level 3

[working_flash_winkfp.zip](https://github.com/uholeschak/ediabaslib/files/4538968/working_flash_winkfp.zip)

uholeschak, 2020-04-27:
Thanks I will check this when I have some time, maybe in one or two weeks.

Martyn-Dev, 2020-04-27:
Thanks, if there is anything you want me to try in the meantime, please let me know.

Martyn-Dev, 2020-04-27:
Donation sent too :)

uholeschak, 2020-04-27:
Thanks very much, I will try to check it as soon possible, but at the moment are also other problems.

terraphantm, 2020-05-03:
Hello, I can confirm that this same issue also occurs with 10MSS54.prg; I suspect the same will hold true for other DS2 modules.

uholeschak, 2020-05-04:
Up to now I had no time to check this.
Which adapter have you used?

Martyn-Dev, 2020-05-04:
I am using a K/DCAN cable flashed with your firmware.

uholeschak, 2020-05-04:
This firmware at least is not supporting telegrams longer than 255 bytes.

Martyn-Dev, 2020-05-04:
I'm not sure that matters as I'm using the same firmware/cable with both your libraries and with plain Ediabas.

uholeschak, 2020-05-04:
When I finally have some time I will check it, but there are more urgent problems a the moment.

uholeschak, 2020-05-13:
I have check the logs now, it seems that the ECU rejects the erase request for some reasons, the transmission error is only a secondary failure. The problem is, that seed_key generates variables telegrams.
Could you create some more log files with EdiabasLib and standard EDIABAS so I could see how the key modifies the telegrams?

Martyn-Dev, 2020-05-13:
Sure thing!

Martyn-Dev, 2020-05-13:
So that's a quick tool32 log containing a 'blocklaenge_max', followed by a 'seed_key', followed by a 'flash_schreiben_ende' with data '010100000000000000FF0000006000600010200300FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF03' and then a 'flash_loeschen' with the same data (data box ticked).

[ifh.trc.working.txt](https://github.com/uholeschak/ediabaslib/files/4622070/ifh.trc.working.txt)

Will get some more logs in a moment.

uholeschak, 2020-05-13:
Thanks, I will check them later.

Martyn-Dev, 2020-05-13:
[api.trc.ediabaslib.api32.log.txt](https://github.com/uholeschak/ediabaslib/files/4622112/api.trc.ediabaslib.api32.log.txt)
[ifh.trc.ediabaslib.api32.log.txt](https://github.com/uholeschak/ediabaslib/files/4622113/ifh.trc.ediabaslib.api32.log.txt)

Logs from using your api32.dll

uholeschak, 2020-05-13:
Could you also add the API trace for ifh.trc.working?

Martyn-Dev, 2020-05-13:
Sorry to be ignorant, but how do i enable API tracing for Tool32 for standard Ediabas?

uholeschak, 2020-05-13:
In working_flash_winkfp.zip you enabled the logging:
API trace level 8 / IFH trace level 3

Martyn-Dev, 2020-05-13:
That was using WinKFP, Tool32 doesnt have the option to enable API tracing.

uholeschak, 2020-05-13:
Could you configure it directly in ediabas.ini?

Martyn-Dev, 2020-05-13:
[api.trc.working.txt](https://github.com/uholeschak/ediabaslib/files/4622264/api.trc.working.txt)

Here you go, it just needed 

ApiTrace = 3

Bunging in ediabas.ini

uholeschak, 2020-05-13:
Thanks, and is this file matching the contents of the old ifh.trc.working.txt?

Martyn-Dev, 2020-05-13:
Yes, same process, same parameters but a different session. I can send you a syncronised log (from same session) if this one is no good.

uholeschak, 2020-05-13:
Please send one from the same session.

Martyn-Dev, 2020-05-13:
Same session.

[ifh.trc.txt](https://github.com/uholeschak/ediabaslib/files/4622289/ifh.trc.txt)
[api.trc.txt](https://github.com/uholeschak/ediabaslib/files/4622290/api.trc.txt)

uholeschak, 2020-05-15:
I think that I have found the reason, the .prg files configures a timeout of 0 while flashing and EDIABAS uses 60 sec. in this case. Are all flashing problems you have related to the DS2 protocol?

Martyn-Dev, 2020-05-15:
I believe so buddy.

uholeschak, 2020-05-15:
Could you check this? I am not sure if I have to modify it also for more protocols.

Martyn-Dev, 2020-05-15:
I experience the same issue when doing the same process on the MSS54 which uses the same DS2 protocol.

uholeschak, 2020-05-15:
Are you able to check if this issue is only related to DS2 protocol and not e.g. BWM-FAST?

Martyn-Dev, 2020-05-15:
I believe it holds true for anything pre KWP2000. @terraphantm may be able to confirm as I believe he is using EdiabasLib successfully with BMW-FAST.

uholeschak, 2020-05-15:
My problem is simply that I am not sure if I could generally set the timeout to 60 sec. if the value is 0.

uholeschak, 2020-05-15:
I have attached a new api32.dll could you check if this fixes the issue?
[Api32.zip](https://github.com/uholeschak/ediabaslib/files/4634530/Api32.zip)

Martyn-Dev, 2020-05-15:
That appears to have worked fine using the updated api32.dll thanks!

Martyn-Dev, 2020-05-15:
WinKFP flash worked fine.

uholeschak, 2020-05-15:
Thanks for the info, at the moment I have modified it only for DS1 and DS2. It would be interesting if other protocols are also involved.

uholeschak, 2020-05-16:
I have added the new timeout handling now also for concept 1 to DS2. I think I will close this ticket now and if you experience problems with other protocols you could reopen it.

Martyn-Dev, 2020-05-16:
Awesome, thank you very much for fixing.

closed 2020-05-16 by Martyn-Dev