uholeschak/ediabaslib #66: Not compatible after flash

strooth, 2020-04-28:
loving this library and managed to get inpa, winkfp, ncsexpert, ista working over wifi with my thor adapter.

I flashed the firmware from the DeepOBD app and now BimmerCode, MHD and others won't work, they can't connect to the Thor Wifi adapter anymore. 

Not sure what to do? or how to fix it? 

I can still get to the esp-link web page etc

I tried uploading the user1.bin and user2.bin files from the library and different pinout settings but I don't know they they should be and can't find any reference, even from Thor themselves. 

I asked them for a firmware to flash back to default but they could not provide me. 

Can you help?

uholeschak, 2020-04-28:
The firmware from Deep OBD is not ELM compatible, it 's working only with EdiabasLib. You could update to the ELM software from the app.

closed 2020-04-28 by uholeschak

strooth, 2020-04-28:
[Image]

I would but its greyed out?