uholeschak/ediabaslib #69: Yandex Translations Are Gone

SyvelDesign, 2020-06-13:
Hi, I just installed DeepOBD today and when I tried to get an API ket for Yandex Translations it says that Yandex doesn't offer free API's anymore. Is there another solution? How do I go about getting a paid API if it's not too expensive?

uholeschak, 2020-06-13:
This is bad but I can't change that, if course you cold pay for it:
https://translate.yandex.com/developers/offer/prices

uholeschak, 2020-06-13:
Maybe this is an alternative:
https://mymemory.translated.net/doc/usagelimits.php

uholeschak, 2020-06-14:
I think the IBM translator is a good solution:
https://cloud.ibm.com/docs/language-translator?topic=language-translator-about

uholeschak, 2020-06-19:
In the beta:
https://play.google.com/apps/testing/de.holeschak.bmw_deep_obd
IBM Watson is integrated.

uholeschak, 2020-06-23:
Released

closed 2020-06-23 by uholeschak