uholeschak/ediabaslib #73: ECUs are missing since the last Deep OBD updates in BMW i3/i8

rs38, 2020-06-27:
e.g. the SME Ecu.
It doesn't show up anymore in the config generator.
I can connect to it with the SME group file but then all the nice values retrieved from "the database" are missing.
Any idea?

uholeschak, 2020-06-27:
Basically the ECU should response to the broadcast command, without more details this is impossible to answer. Please check the trace file.

rs38, 2020-06-27:
I sent the trace file to you via the in-app function. How can you identify the mail?

uholeschak, 2020-06-27:
You have to specify the app id or you have to write a special comment while sending.

rs38, 2020-06-27:
Deep OBD für BMW und VAG
Version: 1.2.256
Id: 4f731353-4de6-442d-91fd-2c625cc7ae69

uholeschak, 2020-06-27:
Sorry there is no trace with this ID. Please specify you email address while sending or add a special comment I could search for.

rs38, 2020-06-29:
I retestet and resend a trace from that device.
It turns out that the global settings checkboxes for "broadcast" and "BWM database" are cleared with every app start. 
Are they supposed not to be saved anymore?

uholeschak, 2020-06-29:
This is a bug, I will check it.

rs38, 2020-06-29:
the same here. it occurs only ~2 of 5 times...

uholeschak, 2020-06-29:
What exactly are you doing to produce the problem?

rs38, 2020-06-29:
I left the car with the settings check boxes enabled and if I now check the app (unplugged) it's unchecked.
There are 2 git commits that touched the settings stuff in the last 6 weeks. It must be there, because it was working fine before.

rs38, 2020-06-29:
> Deep OBD für BMW und VAG
> Version: 1.2.256
> Id: 4f731353-4de6-442d-91fd-2c625cc7ae69

yes 1.2.256 and the problem was already in the version before (but that was probably a beta)

uholeschak, 2020-06-29:
Was it only lost in version 1.2.255? This could be possible.

rs38, 2020-06-29:
no, the bug is in the current version.

uholeschak, 2020-06-29:
Could you create a video who to reproduce it?

rs38, 2020-06-29:
I think it would be much easier to check 1.2.255 again if it works fine. How Do i get it?

pretty sure that
`if (!ActivityCommon.StaticDataInitialized || !_instanceData.GetSettingsCalled)`
is not working as supposed.

uholeschak, 2020-06-29:
This is only the fix, but the problem must be older.

uholeschak, 2020-07-03:
There is now a Beta Version:
https://play.google.com/apps/testing/de.holeschak.bmw_deep_obd
Could you check if this fixes the problem?

rs38, 2020-07-06:
checked new beta 1.2.258:

1.	no cable attached
2.	starting DeepOBD App
3.	goto global settings
4.	make settings changes (broadcast+BMW database)
5.	go back to main menu
6.	check settings again, still checked fine.
7.	exit the App
8.	force the app to close by Android’s App settings
9.	go back to the app, check settings, they are still lost

uholeschak, 2020-07-06:
Sorry I can't reproduce the problem. How did you stop the app? By the force stop button in the system menu?

rs38, 2020-07-06:
yes, as explained via email

uholeschak, 2020-07-06:
Could you create a video?

rs38, 2020-07-06:
no, sorry. don't you believe me???
what could be misunderstood with steps 1.-9. ?

uholeschak, 2020-07-06:
I simply can't reproduce it, it must a special condition or setting that is unknown.
Tried reinstalling the app?

rs38, 2020-07-06:
yes, reinstall was my first try before opening the issue.
let me see if I find another Android device if it behaves differently.

Best bet would be to try an older version with loading an old .apk
can you provide?
I could only narrow down the versions since it occurs, it was in the last 4-6 weeks and according to the commits there were many changes with global settings.

uholeschak, 2020-07-06:
The changes in the global settings are mostly after to problem occurred to fix it.
I expect that this is an indirect effect due to overflow of some memory or similar.

rs38, 2020-07-06:
can you direct me to a trustworthy source of an ~3 month old version of deep OBD .akp?

uholeschak, 2020-07-06:
https://www.dropbox.com/s/25wfq9seu0hb68b/298.apk?dl=0

uholeschak, 2020-07-06:
Jetzt hat mir auch wieder Jemand geschrieben, das die Neuinstallation das Problem behebt.

uholeschak, 2020-07-08:
I think that I have found now the reason for settings failure. A data type has changed that is now incompatible and causes an exception. This is why reinstalling fixes the problem, because all data will be deleted.

closed 2020-07-09 by uholeschak