uholeschak/ediabaslib #75: programming for Deep OBD

thijsdeschildre, 2020-07-14:
I succesfully programmed Binaries-20200517\CanAdapterElm\yc1021\CanAdapterElm.X.production.unified.hex to my adapter with the right PIC and matching bluetooth adapter. MPLab sees the PIC and says programming and verification are good. Bluetooth sees the device still show up as "OBDII" and Deep OBD says "adapter type detection failed".

Am I missing something? I somehow get the impression Deep OBD can program an adapter too, but I can't figure it out from the documentation.

uholeschak, 2020-07-14:
Maybe your PIC firmware type is not matching Bluetooth firmware type.

thijsdeschildre, 2020-07-14:
I'm not sure I understand. My board has the yc1021 bluetooth adapter. So that bluetooth adapter can have different firmware versions, and not all of them all supported?
Should the device connect to Bluetooth as "Deep OBD" or something?

uholeschak, 2020-07-14:
Did you change the yc1021 settings:
https://github.com/uholeschak/ediabaslib/blob/master/docs/Replace_Elm327_BT_Mini_Firmware.md#step1-program-the-yc1021-bt-settings

closed 2020-07-14 by uholeschak