uholeschak/ediabaslib #79: ICOM Nano Support

viktormorozow, 2020-09-18:
Hallo, ich habe ein original ICOM Nano, dies kann ich mit DeepOBD zwar nutzen nach manueller Eingabe der IP (192.168.4.1), die App fragt aber immer wieder ob man nicht das WLAN/Netz wechseln will. Anscheinend wird das ICOM Nano nicht so richtig erkannt, funktioniert aber in der App soweit alles.

Kann man vielleicht den Support für ICOM Nano vervollständigen in DeepOBD/EdiabaLib?

Falls Informationen zum Interface benötigt werden, einfach melden. Für Tests stehe ich auch zur Verfügung.

lg
[image]
[image]

uholeschak, 2020-09-18:
Ja das ist bekannt. Wenn Sie mir einen Adapter zusenden, kann ich das korrekt implementieren, sonst ist das zu aufwendig.

viktormorozow, 2020-09-18:
Das ist schwierig mit Zusenden, nutze den selbst immer wieder. Es funktioniert ja soweit alles, vielleicht muss man ja nur die IP zu den bekannten Geräten hinzufügen?

uholeschak, 2020-09-18:
Nein da ist mehr zu ändern.
Ok, dann scheint es nicht so wichtig zu sein.

closed 2020-09-18 by uholeschak