uholeschak/ediabaslib #83: How to build

elbow, 2020-11-07:
Hi,

I’d like to add feature to Deep OBD to post values to an mqtt broker.  If I’m successful I’ll send a PR.

I’m not experienced with .net development though, to get me started are there some build instructions available? (couldn’t find in the source tree).

Thanks

uholeschak, 2020-11-07:
If you have correctly installed VS with Xamarin simply compile BmwDeepObd.sln

closed 2020-11-07 by uholeschak

elbow, 2020-11-08:
Windows only?

I installed VS on Mac and opened BmwDeepObd.sln and a build gets many errors, eg:

`/Users/steve/ediabaslib/EdiabasLib/SharpZipLibAndroid/CSC: Error CS2001: Source file '/Users/steve/ediabaslib/EdiabasLib/SharpZipLibAndroid/SharpZipLib/src/ICSharpCode.SharpZipLib/Core/Exceptions/StreamUnsupportedException.cs' could not be found. (CS2001) (SharpZipLibAndroid)
`
My SharpZipLibAndroid/SharpZipLib directory does not contain a src library.

This is a git clone of an unmodified fork of your repo.

uholeschak, 2020-11-08:
SharpZipLib is a submodule maybe you have not checked out it correctly ...

elbow, 2020-11-08:
Ah - thank you.

In case it helps anyone else:

# Fetch submodules:
git submodule init
git submodule update

Having done that a build delivers the error:

`/Users/steve/ediabaslib/BmwDeepObd/CSC: Error CS1566: Error reading resource 'BmwDeepObd.HexFiles.Type2.CanAdapterElm.X.production.hex' -- 'Could not find a part of the path "/Users/steve/ediabaslib/EdiabasLib/CanAdapterElm/CanAdapterElm.X/dist/default/production/CanAdapterElm.X.production.hex".' (CS1566) (BmwDeepObd)`

There is no "dist" subdirectory in CanAdapterElm.X directory.

Is it that I need to build EdiabasLib?  Looks like I can't do that successfully on my Mac - so if I need to do that I'll have to find a windows box.

Thanks,