uholeschak/ediabaslib #86: IINPA Fehler bei Livedaten Erfassung

Alpha330i, 2020-11-25:
Ich bekommen nach einigen Minuten Livedaten Erfassung am E60 Bj 2005 plötzlich den Fehler Ifh 0003 Datenübetragung-Hostinterface gestört. 

Adapter: ein von mir selbst programmierter PIC18f25k80 und YC1021 BT Chip, der wiederum auf mit der modified.bin aus den Binarys geflasht wurde. Im Grunde funktioniert dieser ja auch, bis auf die instabile Kommunikation.

Hat jemand das Problem gehabt oder kennt einen Lösungweg?

[image]

uholeschak, 2020-11-25:
Kann man unter Umständen in den Trace etwas erkennen? Es wird schon ein Problem mit der Verbindung bestehen ...

Alpha330i, 2020-11-25:
Leider nicht, gibt es bekannte Probleme mit dem YC1021? 
Könnte der E60 2005 durch eine Pin 7+8 Brücke gestört werden in der Verbindung?

uholeschak, 2020-11-25:
Die Brücke wird es eher nicht sein, Sie können ja mal mit einem anderen Bluetooth Chip testen ...

closed 2020-11-25 by uholeschak