uholeschak/ediabaslib #90: reflashing ELM327 - unknown BT chip

apnof, 2021-01-06:
Hello and thanks to @uholeschak for this projekt!

### Situation
* I've got a ELM327 bluetooth adapter with the PIC18F25K80, see [1] below
* I reflashed the PIC with the replacement firmware (.hex from 'default' folder (`Binaries-20200517\CanAdapterElm\default\CanAdapterElm.X.production.unified.hex`))
  * DeepOBD Android App can connect and read the firmware version
  * Using the App [Serial bluetooth terminal](https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal), I receive an answer from the PIC (adapter powered on desk, no connection to car), see [2] below 
* Flashing was pretty straight forward thanks to the exposed PIC pins on the board, I labeled them if anyone else got the same adapter: 
[image]

### Problem/Question
**How to identify the bluetooth chip on my board?**
As stated on [Replacement firmware for ELM327](https://github.com/uholeschak/ediabaslib/blob/master/docs/Replacement_firmware_for_ELM327.md) it is recommended to flash the bluetooth chip in addition to the PIC FW as well to get 115200 baud, the possibility to change the bluetooth name and maybe improved stability (?) (and of course use the matching PIC FW instead of the 'default' one)

#### Observations regarding my BT chip: 
* 32 pin package
* no visible text on chip (looks like polished off - rough surface, very sensitive - gets easily scratched by fingernails)
* does not have an external EEPROM for the configuration data like the yc1021.
* other pins for the external clock (XTAL) compared to the BK3231 
* Does the BC417 always come with a bga package? Then it is no BC417
[image]

### some details
[1]
[image]

[2]
```15:01:53.668 Connecting to OBDII ...
15:02:13.320 Connected
15:02:58.864 82 F1 F1 FE FE 60 0D 0A
15:02:59.016 82 F1 F1 FE FE 60 82 F1 F1 FE 81 E3
15:03:03.869 Disconnected
```

uholeschak, 2021-01-06:
At least on the photo I see no BC417 ...

closed 2021-01-06 by uholeschak

apnof, 2021-01-06:
But doesn't the YC1021 always need a configuration eeprom? 
There are no chips on the other side of the pcb. The other 8-pin chip on the left side seems to be supply-voltage related..
So it doesn't look familiar to you? Then I will cancel and stay on the 'default' firmware with bluetooth name 'OBD II' for now.

I modified another adapter a few weeks ago and this one had a TC1021 with an additional eeprom (so the baudrate could be changed and I used the PIC FW from the corresponding folder of the binary package)
[image]
Ordered the second elm from the same seller on aliexpress and of course, I received something different :roll_eyes:

If anyone else got a hint for me feel free to point me in the right direction.

uholeschak, 2021-01-06:
At least it should be the unknown chip on the right side of the first picture.

Alpha330i, 2021-01-08:
It is the BK3231 Chip like uholeschak said. There is also a possibility to flash the bc04 firmware onto it but i dont know exactly the procedure.

uholeschak, 2021-01-08:
Yes the BK3231 seems logical, but it's ARM based and I have I found no way to update it.

Alpha330i, 2021-01-08:
Maybe this topic could be useful in future, if someone is interested into flashing BK3231 / JDY-30

uholeschak, 2021-01-08:
Via the SPI interface I was already able to program the device, but this is only possible with the large housing, the small one has no SPI. There the flash interface is required. There are commercial flashers for this device, but they are too expensive.

Alpha330i, 2021-01-08:
Okay, that sounds very interesting, thank you for this hint. Is the the BC04 firmware included into your sourcecode libary as a .bin file? I found a BK3231 Flash folder but i think it was just for your internal tests

uholeschak, 2021-01-08:
Sorry no, for this I have to be able to read the device, not possible without flasher and the device probably locked.

apnof, 2021-01-08:
Sorry, I'm stuck:

Where is the difference between
`Binaries-20200517\CanAdapterElm\bc04\CanAdapterElm.X.production.unified.hex`
and
`Binaries-20200517\CanAdapterElm\default\CanAdapterElm.X.production.unified.hex`
?

When there is no ediabaslib specific firmware for the BK3231 why the PIC firmware in the bc04 folder (checksum of hex files differ)?

Alpha330i, 2021-01-08:
The BC04 Firmware is on the BK3231 Chip, and the BC04 Folder with the Pic Firmware is only for the matching BC04 on BK3231.
If The BT Modul (BK3231) has no BC04 Firmware by factory, then you have to use the default Pic Firmware. 
BC04 Pic and Default Pic Firmware are not the same, Uholeschak found a way to use the BC04 Firmware (if it is by factory on BT Chip) and made it able to work at 115200 baud + Pin / Name changeable.
But its not possible to flash BC04 on existing BK3231 when its not the factory firmware type

uholeschak, 2021-01-08:
Yes the BC04 will be programmed to use a higher baud rate.

apnof, 2021-01-11:
Thanks for the explanations! 😊

---
In conclusion (and to check if I understood now):
* There is a serial connection between the PIC microcontroller and the bluetooth chip (obviously)
* there are multiple bluetooth chips on the ELM adapters (with PIC18F25K80 in out case), here is a listing/description of the chips: [Replacement firmware for ELM327 | Programming ELM327 adapter with Deep OBD firmware](https://github.com/uholeschak/ediabaslib/blob/master/docs/Replacement_firmware_for_ELM327.md#programming-elm327-adapter-with-deep-obd-firmware)
* normally, the elm327 firmware uses a 38400 baud connection between PIC and BT chip, so the BT chips are configured this way (either within their firmware or with an external configuraion eeprom (e.g. the YC1021 uses that))
  * for unmodified/unconfigured BT chips, the 'default' PIC firmware is used (38400 baud)
* ideally, for DeepOBD firmware, we want to use 115200 baud, so there are multiple subfolders for the different BT chips in the binary package
* in the case of BK3221, there are variants
  * which come with a custom firmware - at this time, we have no possibility to reconfigure or reflash it (except with an expensive commercial flasher)
  * which are being shipped with the so called BC04 firmware. This firmware can be reconfigured(?) to use 115200 baud [Question a)]
---
* Did I understand correctly?
* Question a) regarding the last bulletpoint: how can the BC04 be reconfigured? Does it communicate with 115200 already and the ELM is using 115200 in these ELM adapter per default as well? I found the page [Custom Bluetooth firmware](https://github.com/uholeschak/ediabaslib/blob/master/docs/Custom_Bluetooth_firmware.md) but it is for HC-04, HC-05, HC-06 on BC417 chips. 

Thanks folks

uholeschak, 2021-01-11:
Yes this is basically correct. The BC04 is configured automatically at startup if it has default settings.

Alpha330i, 2021-01-11:
But to be honest, i didn't notice the difference between baud 38400 & 115200 yet. Maybe its noticeable while data logging, but for my normal use not.
I think the Bk3231 with bc04 can be configured principel with AT command simmilar to BC417 with hc04?

uholeschak, 2021-01-11:
Yes with AT commands.

GyurkanM, 2021-01-18:
@apnof Did you manage to identify the BT chip on this PCB?

apnof, 2021-01-20:
Not in-depth.. I went with the assumption that it is a BK3231.
At least the Antenna output matches the datasheet :sunglasses:

[image][image]

harryberlin, 2021-01-20:
have a look at the crystal traces.