uholeschak/ediabaslib #95: Is there any chance to use this diagnostic tool for BMW motorcycle?

0xFACE, 2021-04-04:
Sorry, this is not an issue, but a beginner's question. 
Is there any chance to use this diagnostic tool for BMW motorcycle?

uholeschak, 2021-04-04:
Perhaps, you have to try.

0xFACE, 2021-04-05:
I have a motorcycle alarm module (DWA) on the desk.
I can connect to it this way:
MOTOSCAN Software <-> Bluetooth <-> ELM 327 <-> CAN-BUS <-> ECU
Most likely, it uses UDS protocol.
Unfortunately, I do not know which ECU to choose in Deep ODB.

uholeschak, 2021-04-05:
What about scanning with the configuration generator?

mi-hol, 2021-05-03:
scanned with configuration generator but error was reported. Note: MotoScan works with the same ELM327 Version 1.5 adapter.

uholeschak, 2021-05-03:
With ELM the communication is only possible if D-CAN is used.

mi-hol, 2021-05-03:
any suggestion how I can get over this hurdle?

uholeschak, 2021-05-03:
First you have to find out which protocol is used by the ECU.

uholeschak, 2021-05-03:
You could flash the standard Deep OBD replacement firmware in the PIC and try again (with PCIKit).

mi-hol, 2021-05-03:
ECU is a Bosch Motronic MA 2.4, does this help?

uholeschak, 2021-05-03:
No

0xFACE, 2021-05-03:
I tested DWA and KOMBI ECU with STN1170 adapter (generic ELM-327 also works with some limitation).
I was able to communicate with UDS over CAN protocol. 
As far as I know, some engine ECU (MOT) uses UDS over K-line as well.
Depends on year and model.

uholeschak, 2021-05-03:
I expect it will work with up to date models.

mi-hol, 2021-06-12:
Please reopen issue.
Attached log file from running DeepODB on BMW R1100S
[BMW_R1100S_1998.zip](https://github.com/uholeschak/ediabaslib/files/6642819/BMW_R1100S_1998.zip)

mi-hol, 2023-08-24:
Does the comment "Keine Antwort vom Fahrzeug.\nBei Motorrädern sollten Sie die Motorräder erkennen Option im Menü aktivieren.\nBei K-LINE Fahrzeugen wird ein USB oder Deep OBD Adapter benötigt." mean I should retest with  [Version 1.3.174](https://github.com/uholeschak/ediabaslib/commit/b09f76c069ce119e1ff163ba18a51ad3294a462d) or later?

uholeschak, 2023-08-24:
If have only added the detection code from ISTA but it's not tested.
You have to enable the extra check box in the menu every time (slow!).
It will only work with motorcycles of board net type 2000 and 2020.

mi-hol, 2023-08-26:
In case this should evolve into a working solution I'd suggest to re-open this issue to attract testers

uholeschak, 2023-08-26:
Ok, but I expect your motorbike is too old.

uholeschak, 2023-10-17:
I have now uploaded a beta version:
https://play.google.com/apps/testing/de.holeschak.bmw_deep_obd
In this version only a single ECU is added (hopefully), if no group SGDB is found.
At least this should help for single ECU vehicle, but I am not sure if this is working.
Could you test it and send a trace file from the configuration generator (APP ID required)?

mi-hol, 2023-10-24:
Hallo ulrich,

Leider bin ich noch nicht zum testen gekommen, da erst der Geschirrspüler und jetzt die Heizung spinnen.

Habs aber auf der (langen) ToDo List

Sorry Micha

 
 

Gesendet: Dienstag, 17. Oktober 2023 um 10:29 Uhr

 

https://play.google.com/apps/testing/de.holeschak.bmw_deep_obd

uholeschak, 2023-10-25:
Das wird schon einige Update-Runden benötigen, bis es richtig funktioniert. Etwas Zeit wird man da schon investieren müssen.

closed 2023-11-05 by uholeschak

uholeschak, 2023-11-05:
Please create a new issue for every motorbike model if there are still problems.