uholeschak/ediabaslib #97: KW902 with PIC18F25K80 and BK3231S

HeadCrash66, 2021-04-10:
Hi,

I bought the following adapter and wanted to flash the alternative firmware.
The flashing into the PIC18F25K80 worked but there seems to be problems with the Bluetooth IC BK3231S.

I liked the adapter because of the shape, if someone has similar plans may take this as warning.
Or if someone has an idea how to get it to work anyway I would be happy.

Best regards.

uholeschak, 2021-04-10:
If you use the default firmware you have the reduced baud rate but no problem.

closed 2021-04-10 by uholeschak

Alpha330i, 2021-04-11:
Ich habe mich vor kurzem an einem BC417 Chip probiert und festgestellt, dass mein bestellter Chip nur 6mb flash size hat. Nun erhalte ich die Nachricht beim flashen, dass das Image verkleinert wird, allerdings funktioniert das ganze dann nicht mit der spp_uart hex Datei auf dem PIC18. Die Kompakte Version funktioniert auch nicht. Gibt es dazu einen Workaround oder muss ich einen neues HC-05 Board kaufen in der Hoffnung ein besseren Chip zu erhalten?

uholeschak, 2021-04-11:
You have to specify compact firmware:
[image]

Alpha330i, 2021-04-11:
which software are you using? maybe that could be a solution. 
There where compact files in your binaries with 4,4mb flash size but they didn't worked

And with the 6,2mb files im getting this message:
[image]

uholeschak, 2021-04-11:
You have to use BlueLab 4.1

Alpha330i, 2021-04-11:
Is there an official download of BlueLab 4.1?  i don't find serious downloads

uholeschak, 2021-04-11:
No the the product is not existing any more.

HeadCrash66, 2021-04-11:
> 
> 
> If you use the default firmware you have the reduced baud rate but no problem.

Many thanks for your reply.
Unfortunately, the default firmware does not work. DeepODB can not access my Konnwei KW902.
With windows I can see two serial ports, so BT is still working, but none of them sends any reply.
I have checked the flash process with verify so I'm pretty sure that it was successful.

uholeschak, 2021-04-11:
Maybe you have to analyze the traffic with an oscilloscope. Wrong baud rate?

Alpha330i, 2021-04-11:
Thank you for your help, now it worked with a pullup resistor on tx line.

uholeschak, 2021-04-11:
Yes for the compact firmware the pullup is required!

Alpha330i, 2021-04-11:
Now the app recognizes that the adapter is connected (changing the Bluetooth name works), but as soon as I connect to the vehicle, no data is displayed, despite the correct config file. I also sent the trace file from "bensaf@outlook.de"

uholeschak, 2021-04-11:
Have used the configuration generator?

Alpha330i, 2021-04-11:
yes i used the configuration generator, but it wasn't able to read out the car (2004 E46). With other OBD Adapter which was bought from you, all is working fine. So i guess its something with the BT chip? ive attached some photos of the Adapter and what is shown on the app. The Hex File is also working fine as you can see on the first picture
[image]
[image]

uholeschak, 2021-04-11:
I expect you have to find the reason yourself ...

Alpha330i, 2021-04-11:
Hmm i think those "fake" BC417 are not working correct, but anyway, i will try another one

uholeschak, 2021-04-11:
Use an oscilloscope an analyze it ...