uholeschak/ediabaslib #99: Does Bluetooth ELM327 replacement firmware work with old E36 witch needs ADS ?

OrpheeGT, 2021-06-29:
Hello,

I have ISTA+ working with a E39 2003 530i car. everything works.

But on a E36 1996 328i car, the VIN is detected, but communication with module (DME) "is pertubed"
I can't access to it. Can't read module infos (serial number etc...)

I'm using the PIN 7-8 connector.

Everything works OK with ICOM unit so I don't think the issue is from the car.

uholeschak, 2021-06-29:
This could be also a connection problem. Ar you using a Deep OBD Bluetooth Adapter?
You could add these entries to the EdiabasLib.config file an check the result in ISTA-D\Ediabas\BIN\Trace
```
    <add key="LockTrace" value="1" />
    <add key="ApiTrace" value="1" />
    <add key="IfhTrace" value="2" />
    <add key="AppendTrace" value="1"/>
    <add key="ApiTraceName" value="api.log"/>
    <add key="IfhTraceName" value="ifh.log"/>
```

OrpheeGT, 2021-06-29:
Yes I'm using a Deep OBD bluetooth adapter.

Do I have to change the Interface value ?

Currently set to STB:OBD

uholeschak, 2021-06-29:
No this is ok, the configuration generated by the config utility should be ok.

OrpheeGT, 2021-06-29:
OK, because with a standard ADS cable, the settings must be set to ADS instead of STD:OBD.

handmade0octopus, 2021-06-29:
Sorry to bodge in but my eye was caught by your car.

I created this library mainly for e36/e39 and since its dorect ds2 protocol its much faster (up to 33HZ) and have many more parameters than standard obd2.

You may find it useful.

https://github.com/handmade0octopus/ds2

OrpheeGT, 2021-06-29:
Here are the logs

[api.log](https://github.com/uholeschak/ediabaslib/files/6732601/api.log)
[ifh.log](https://github.com/uholeschak/ediabaslib/files/6732602/ifh.log)

[image]
[image]

@handmade0octopus I will have a look but I'm just an advanced end user...

Thanks

uholeschak, 2021-06-29:
Yes in the most cases there is no response, but I can't tell you why. Is it working with the app?

OrpheeGT, 2021-06-29:
So I have some news.
I just received CH341A flasher https://www.amazon.fr/gp/product/B08J2PPHDY
to flash YC1021 bluetooth chipset.
You may add it in the list as it is really cheaper than the 2 currently listed.

So I flashed modified.bin, and flashed YC Deep OBD FW variant. And it all works.

Regarding ISTA+ still the same issue... But I tried my default INPA installation on same laptop (just had to patch default Ediabas)

And INPA works, it detect the ignition, and I can access DME (ID, serial, error memory...)

I'm using ISTA+ 4.29.30 ...

uholeschak, 2021-06-29:
If the app is not able to read the vehicle database is scans the complete ECU list, this requires some time.

uholeschak, 2021-06-29:
If you are able to create a log with standard Ediabas i could compare ...

OrpheeGT, 2021-06-29:
I will do in a few hours (back at work)

OrpheeGT, 2021-06-29:
Hi,

So I enabled logs on INPA too.

I ran it with INPA and with ISTA on both E36 and E39

[ISTA_Trace.zip](https://github.com/uholeschak/ediabaslib/files/6734591/ISTA_Trace.zip)
[INPA_Trace.zip](https://github.com/uholeschak/ediabaslib/files/6734585/INPA_Trace.zip)

On the E39, I managed to have a timeout.
Inside a module, reading some values, and then I had a "communication perturbée"

But most of tests I made worked (cluster niddles and light test, serial numbers etc...)

With K+Dcan cable with usually set COM port to 1ms maybe something to look at ?

Thanks

uholeschak, 2021-06-29:
Ok thank, I will try to check the files, but comparing will get difficult. I will try when I have some time.

OrpheeGT, 2021-06-29:
On E36 logs
 There is a lot of "*** No header received" on ISTA+ logs.
Then : 

SetError: EDIABAS_IFH_0009
executeJob Exception: Error occured: EDIABAS_IFH_0009
executeIdentJob Exception: executeJob
Error occured: EDIABAS_IFH_0009

Or also : 

SetError: EDIABAS_IFH_0009
executeJob Exception: Error occured: EDIABAS_IFH_0009
executeJob('tmode.prg'): 'setze_interface_zurueck'
executeJob Exception: executeJob: Function not implemented

Or : 

SetError: EDIABAS_IFH_0006
executeJob Exception: Error occured: EDIABAS_IFH_0006
 (ArgBinary): FF FF 00 00 
executeJob('tmode.prg'): 'SETZE_ANTWORTLAENGE'
Answer length: FFFF 0000
 (ArgBinary): 12 04 00 
executeJob('tmode.prg'): 'SENDE_TELEGR_WIEDERHOLT'
 (Send Frequent): 12 04 00 
SetError: EDIABAS_IFH_0006
executeJob Exception: Error occured: EDIABAS_IFH_0006
executeJob('tmode.prg'): 'STOPPE_WIEDERH_ANFORDERUNG'

None of them appears in INPA traces.

ISTA+ execute job : executeJob('tmode.prg') very often

I also have theses error on E39 ISTA+ logs :

 (Send): 83 10 F1 22 F1 50 E7 
*** No header received
 (Send): 83 10 F1 22 F1 50 E7 
*** No header received
SetError: EDIABAS_IFH_0009
executeJob Exception: Error occured: EDIABAS_IFH_0009
executeIdentJob Exception: executeJob
Error occured: EDIABAS_IFH_0009

The following error happens also :

executeJob('lcm_iv.prg'): 'C_FA_LESEN'
SetError: EDIABAS_SYS_0008

I can see the moment when ISTA+ scan all E39 modules and there is no error in this period except for SES module.

It can't find SES module as I removed it and replaced with a BlueBus module so it is expected

Then sometime after :

Start comm thread
 (Send): 80 05 04 01 80 
 (Resp): 80 14 A0 01 90 84 25 00 00 00 00 00 00 00 00 00 00 00 00 04 
executeJob('ms430ds0.prg'): 'IS_LESEN'
SetError: EDIABAS_SYS_0008
executeJob('gs8604.prg'): 'IS_LESEN'
SetError: EDIABAS_SYS_0008
executeJob('ews3.prg'): 'IS_LESEN'
SetError: EDIABAS_SYS_0008
executeJob('mfl2.prg'): 'IS_LESEN'
SetError: EDIABAS_SYS_0008
executeJob('dsc57.prg'): 'IS_LESEN'
SetError: EDIABAS_SYS_0008
executeJob('lws5.prg'): 'IS_LESEN'
ReadAllJobs file not found: C:\EC-APPS\ISTA\Ecu\_lws5.prg
SetError: EDIABAS_SYS_0008
executeJob('ihka39_5.prg'): 'IS_LESEN'
SetError: EDIABAS_SYS_0008
executeJob('pdcact.prg'): 'IS_LESEN'
SetError: EDIABAS_SYS_0008
executeJob('radio.prg'): 'IS_LESEN'
SetError: EDIABAS_SYS_0008
executeJob('dsp2.prg'): 'IS_LESEN'
SetError: EDIABAS_SYS_0008
executeJob('navmk4_2.prg'): 'IS_LESEN'
executeJob('navmk4_2.prg'): 'INITIALISIERUNG'
StopFrequent
Stop comm thread
Frequent mode stopped
Stop comm thread

Then I can also see the same error found for E36 :

executeJob('lcm_iv.prg'): 'IS_LESEN_DETAIL'
SetError: EDIABAS_SYS_0008
executeJob('tmode.prg'): 'setze_interface_zurueck'
executeJob('tmode.prg'): 'INITIALISIERUNG'
executeJob Exception: executeJob: Function not implemented

It it in my opinion the main differences I found...

[image]
[image]

uholeschak, 2021-06-29:
Thank good hint:
executeJob Exception: executeJob: Function not implemented
Nom du paramètre : xboot
This may the the problem ...

OrpheeGT, 2021-06-30:
Is there anything I can do more to help you ?

uholeschak, 2021-06-30:
Thank, I will create a new version with the xboot command, then you could test again ...

OrpheeGT, 2021-07-01:
Just a quick question :
Does Deep OBD FW / EdiabasLib use L-Line (ADS) on PIN15 for old E36/E34 cars ? Or does it work only On K-Line (17/20 <-> 7/8)
My 20 pin round adapter currently don't has the Pin15 <-> Pin15 wired.
Thanks

uholeschak, 2021-07-02:
If you use the Deep OBD BT Adapter it will be used. I think also with FTDI USB

OrpheeGT, 2021-07-02:
From my knowledge PIN15 L-Line is only used by old ADS interface with serial port. K+DCan USB cable never used it and so you were not able to access all modules from an ADS PIN15 car with a USB cable. 

I will have to rebuild a 20Pin round adapter with PIN15 (and PIN16 ignition) active.

Thanks

uholeschak, 2021-07-02:
I remember well, the typical DS2 protocol is nut using L line.

OrpheeGT, 2021-07-02:
Yes, if I'm not mistaken : 
DS1 = ADS mode
DS2 = OBD mode

Does Deep OBD able handle DS1 ?

uholeschak, 2021-07-02:
In EdInterfaceObd.cs
                    case 0x0005:    // DS1
                    case 0x0006:    // DS2
seems to be present ...

OrpheeGT, 2021-07-13:
Hi, 
So you made a great work on this past days, maybe the ticket can be closed ? almost all were addressed it seems ?

Thank you again.

uholeschak, 2021-07-13:
I expect we we search more we will find other things, but for the moment we could close it.

closed 2021-07-13 by uholeschak