uholeschak/ediabaslib #103: K+DCAN cable update

SantCineva, 2021-11-07:
I've come across this tutorial ( http://blog.obdii365.com/2021/07/09/update-inpa-kdcan-cable-with-enhanced-firmware/ ) and I wanted to know if it would work  on a normal K+DCAN cable like the ones in the tutorial without modifiing anything else.
Should I just update the flash and fuses and expect everything to work?
I'm mostly using DIS software and INPA.

uholeschak, 2021-11-07:
I think you should ask this the creator of the tutorial.

closed 2021-11-07 by uholeschak

SantCineva, 2021-11-07:
What do you mean? Can't you clarify this?

uholeschak, 2021-11-07:
This tutorial is not from me, so I can't tell you anything about it.
http://blog.obdii365.com/2021/07/09/update-inpa-kdcan-cable-with-enhanced-firmware/

SantCineva, 2021-11-09:
I tested it myself on a green PCB, K+DCAN adapter, and it works .
I thought it might be a good idea to leave some feedback here. I've seen a lot of people having the same question on a fiew forums.
Is there any chance I could get the source code of the file "Combined.hex"? Or at least some documentation on it?

uholeschak, 2021-11-09:
Its all open source:
https://github.com/uholeschak/ediabaslib/tree/master/EdiabasLib/CanAdapter