uholeschak/ediabaslib #106: Additive trim(short term fuel trim) and rough running values incorrect on N42B20A engine(ME9K_NG4.prg)

GyurkanM, 2021-11-26:
I have submitted trace from thepcmaniacc@gmail.com yesterday, as you can see STATUS_GEMISCH is read 3000 which is impossible value. The job used by inpa is STAT_ADD_WERT, but when I choose it from prg and read it again it is read 3000 which is still incorrect value, so I suppose there is bug in EdiabasLib. Rough running values should be read from STATUS_MOTORLAUFUNRUHE, but in BMW database it is from STATUS_LAUFUNRUHE which is wrong , shows implausible values 0-180, but its ok when I select from the prg file STATUS_MOTORLAUFUNRUHE job. I append xml files generated with xtract.exe from the ME9K_NG4.prg for your reference, and the decompiled IPO file which is used by INPA also, as I think it can be of use to you.
[files.zip](https://github.com/uholeschak/ediabaslib/files/7610527/files.zip)

uholeschak, 2021-11-26:
Could you test it with tool32 and and compare the results of EdiabasLib and Ediabas output? I expect more that this as problem in the .prg file.

GyurkanM, 2021-11-26:
Yes have done that, when using Ediabas STAT_ADD_WERT is read correctly, with EdiabasLib the value is incorrect no matter if it is read on DeepOBD or Tools32/Inpa/Ista on PC. What files will be of use to you so you can understand what is going on? Thanks!

uholeschak, 2021-11-26:
At least I have found your app trace files. Do you also have api and ifh traces from tool32 (both cases?)

GyurkanM, 2021-11-26:
No at the moment, will generate traces for the both cases, thanks!

GyurkanM, 2021-11-28:
Hi, I have generated api/ifh traces with both tool32/inpa for ediabas and ediabaslib and attach them
[traces.zip](https://github.com/uholeschak/ediabaslib/files/7613098/traces.zip)
.

uholeschak, 2021-11-28:
Thanks, which value I have to check? There is no STAT_ADD_WERT in the api traces.

GyurkanM, 2021-11-28:
STATUS_GEMISCH in this case, eg shows same results as STAT_ADD_WERT

uholeschak, 2021-11-28:
The result STAT_RKAT_WERT?

GyurkanM, 2021-11-28:
Yes, updated the comment above

uholeschak, 2021-11-28:
Thanks, I will check that when I have some time.

GyurkanM, 2021-11-30:
Thanks for your efforts! Can I somehow fix this in the DeepObd app without waiting for the next update of the app?

uholeschak, 2021-11-30:
No I could create a beta in the next days.

uholeschak, 2021-12-01:
Here is a beat version:
https://play.google.com/apps/testing/de.holeschak.bmw_deep_obd

uholeschak, 2025-09-28:
Where I have told that I have decompiled `.ipo` files? I have only decompiled `.prg` files.