uholeschak/ediabaslib #110: Deep ODB installation endless loops in "extracting files" as soon as screen lock was activated

mi-hol, 2022-01-26:
Android phones have usually a time set to lock automatically.
When Deep ODB "extract files" is running at the time screen lock gets activated, the process seems to stop.
At least when phone is unlocked, Deep ODB displays the "Question" dialog an ask for yes/no confirmation to extract 2.3GB ECU files again.
This results in an endless loop and installation never finishes.
Installation should correctly handle "device lock" and "device unlock" events and resume gracefully.

uholeschak, 2022-01-26:
I expect the app will terminate. The only way would be preventing the lock.

mi-hol, 2022-01-26:
> preventing the lock.

correct, this approach is used by video players, navigation apps.
Maybe the source code of vlc for android contains the solution.

uholeschak, 2022-01-26:
At the moment I have only kept the CPU active, change it to display dim. Have to test it when I have time.

mi-hol, 2022-01-27:
In case a .apk file is downloadable, I'd be happy to test

uholeschak, 2022-01-27:
I have uploaded Version 1.3.60
https://play.google.com/apps/testing/de.holeschak.bmw_deep_obd

mi-hol, 2022-01-28:
Great news, the change fixed the issue!
Thanks you so much.

closed 2022-01-28 by mi-hol

uholeschak, 2022-01-28:
Thanks for the feedback.