uholeschak/ediabaslib #117: Can not re-flash with elm327 firmware

GyurkanM, 2022-03-19:
After the last beta update or the previous one function to re-flash with elm327 through the deebobd is not working. Adapter is flashed with yc1021 firm and previously have changed between elm327 and ediabaslib many times.

uholeschak, 2022-03-19:
Basically nothing has changed here. What exactly is the problem? Which update exactly cause the problem?

GyurkanM, 2022-03-19:
I should try by reainstalling the app maybe, but also I changed the phone from lenovo p2 to galaxy s21, maybe this is the issue, will check with the old phone, also with the regular release, so I can pinpoint the exact problem.

uholeschak, 2022-03-19:
I am also using a Galaxy S21, maybe I could check it again ...

GyurkanM, 2022-03-19:
The issue resolved after several times re-pairing the adapter.

closed 2022-03-19 by GyurkanM

uholeschak, 2022-03-20:
Thanks for the info, so probably this was a connection problem.