uholeschak/ediabaslib #118: [Questino] UDS Response payload from 0x22 request

Matheus-Garbelini, 2022-04-01:
Hello, I'm trying to understand how to extract response payload from the UDS response.

For instance:
When I send UDS ReadDataByCommonIdentifier {0x22,0x25,0x04}, 
The ECU replies ({0x62, 0x25, 0x04, **0x00**}, where 0x00 is the actually data from the UDS positive response.

How do we map the response to the results declared in the prg? For instance, in the prg, it indicates that I can have multiple response but the decoding of this is not clear.

Attached the extracted UDS jobs information I got from jbbf3 ecu. Note that it what is expected from the responses but it doesn't say what is the correct order of elements.

[jbbf3.txt](https://github.com/uholeschak/ediabaslib/files/8396445/jbbf3.txt)

uholeschak, 2022-04-01:
If you wan to understand how the .prg file is analyzing the result, you have to decompile it. There is no "mapping".

closed 2022-04-01 by uholeschak

Matheus-Garbelini, 2022-04-01:
@uholeschak thanks.

Is this what your tool/FileDecoder does?
Just want to understand what part of this repo parses the .prg file so I can create a scaly conversion tool for the parsed/decompiled output.

Regards.

uholeschak, 2022-04-01:
Probably decompiling will not help you. Simply read the UDS specification and implement it it yourself ...

Matheus-Garbelini, 2022-04-01:
Hi @uholeschak, thanks for your prompt response.
I understand that UDS spec will not help here because the service response payload is prorietary.

I'm referering to this feature of this library "The library can directly read and interpret .GRP and .PRG files"
One of the arguments of [EdiabasTest](https://github.com/uholeschak/ediabaslib/blob/master/docs/EdiabasTest_parameters.md) receives the prg file argument to execute a job. 
**My question is: Does this EdiabasTest parses the response of the job or it just sends a formated request taken from the prg?**

I realized that my question could be taken in bad faith or misunderstood.
* I'm not looking to create any sort of competitor tool so implement myself is not feasible. This is a complex reverse engineering work and thanks for your great work on this open-source library.
* I'm just looking to **extend** this existing library to create python scapy layers for automated BMW network research and intrusion detection. I just wanted to understand the high-level of **where** prgs files are handled here as it seems BmwDeepObd parses the response after a configuration file is generated (from prgs to custom xml). 
* Specifics on how to integrate your C# code with python code I can figure it out later without bottering you.

Thanks for your patience.

uholeschak, 2022-04-01:
EdiabasLib simply executes the assembly code in the .prg file. It has no understanding of the response interpretation.