uholeschak/ediabaslib #119: Unable to find ECU

2no1, 2022-04-29:
Group: VAG
Engine: 1.9 TDI
Engine code: AHU

adapter is configured for automatic protocol selection.

the device is unable to connect to the ECU. 
the progress bar wont even move - goes straight from 0/100 to unable to connect. this will happen even with the car running.

i tried switching to the BMW group and when Detect ECU is selected the progress bar moves and an output is shown.

am i skipping something i need to do before detecting the ECU?

uholeschak, 2022-04-29:
Is the vehicle old enough? Support is only until 8.2017
Do you get an error message or it's imply hanging?
If there is no support an error would be ok.

2no1, 2022-04-29:
> Is the vehicle old enough? 
1997
>Do you get an error message or it's simply hanging?
the error is "unable to detect ECU". i guess it sort of hangs in the 0/100 screen and it immediately jumps to the "unable to detect ecu" screen

uholeschak, 2022-04-29:
If the monitor is not detected it will abort. Which protocol is used?
Have you send a trace file? The App ID is required.

2no1, 2022-04-29:
im sorry, i was mistaken.
the error is "No response from vehicle"

from what i can find online, the protocol is KW 1281
yes, using Deep OBD without 7+8 pin connection.
just sent a trace file. what app ID?

uholeschak, 2022-04-29:
Please send me the app ID from the main menu.

2no1, 2022-04-29:
> Please send me the app ID from the main menu.

Id: 5bdc0c31-9080-4dc5-a90b-1261df0b8538

uholeschak, 2022-04-29:
Thanks, I could only see that there is no response while baud rate detection.

2no1, 2022-04-29:
what could this mean? is it an adapter issue? ECU issue?
i have just sent another trace file when i was asked if i wanted to send one.

closed 2022-04-29 by uholeschak