uholeschak/ediabaslib #121: Unable to read PIC18F25K80 using PICkit 3 to flash ELM327 firmware

chaddoncooper, 2022-05-07:
Apologies, I am not a PIC user, so might be missing something obvious. But I can't seem to get the chip to read on my ELM adapter.

So I purchased the same adapter as here and hooked it up the same (10k SMD Resistor):
https://github.com/uholeschak/ediabaslib/issues/39#issuecomment-623412265

Followed the steps in the guide W11/Admin Mode, to install PICkit 3 with the mods up to ["Try to read hex file from PIC18F25K80. Do not continue until you get a proper hex file"](https://github.com/uholeschak/ediabaslib/blob/master/docs/Replace_ELM327_HC04_Firmware.md#step2-program-pic18f25k80).

Alas all I'm getting is a blank hex file. I'm not sure if the Code Protect has anything to do with it or if I need any other options checked? Blank check returns that the chip is not blank.

[Screenshot of read in Pickit 3](https://i.imgur.com/lxQrNAE.png)

Wiring seems ok and lights seem to flash in the correct order when I try a read:

[Image of Read](https://i.imgur.com/Cv25Llv.jpg)
[Video of Read](https://www.youtube.com/watch?v=roaqXmII8Is)

Any ideas on what might be the issue?

uholeschak, 2022-05-08:
Make sure the lines are corrected correctly and are not influenced by the adapter hardware.

closed 2022-05-08 by uholeschak

chaddoncooper, 2022-05-23:
Soldering was fine. Found my issue - there's a menu item within the Pickit 3 software which you simply untick to disable the code protect. May be helpful for anyone else unfamiliar with this software/ecosystem. Sadly I only noticed after taking the chip off the board to accomplish the flash.