uholeschak/ediabaslib #124: "UniCarScan UCSI 2100" Adapter Support

PVC-C, 2022-06-16:
Hello,

is it possible to use the Adapter [UCSI2100](https://www.wgsoft.de/unicarscan-ucsi-2100) with K-Line Vehicles?
It is detected as not compatible with K-Line Communication but is working fine with it.
Could it just be whitelisted or does it need coding?

Thanks for this great Piece of Software.

uholeschak, 2022-06-16:
For K-Line vehicles only Deep OBD and FTDI adapters are working, no ELM327.
https://github.com/uholeschak/ediabaslib/blob/master/docs/AdapterTypes.md

closed 2022-06-16 by uholeschak

PVC-C, 2022-06-16:
Deep OBD Firmware "just" adds K-Line support to ELM327?
The Adapter already has K-Line support with its firmware.

uholeschak, 2022-06-16:
The Deep OBD firmware is absolutely no ELM firmware.
ELM327 K-Line support is not compatible with protocol the app is using.