uholeschak/ediabaslib #126: More information on source code

johann74270, 2022-07-12:
Hello,

Thank you for this extremely complete application, you did a great job.

Being thirsty for knowledge, I would like to have more information on how your programming classes work.
I'm learning IT development

Would it be possible to comment more on the source code that you propose or edit a documentation file on github to explain your source code?

Thank you and best regards

mark-sch, 2022-07-12:
Look into the amount of commits. Ulrich is doing tons of quality coding ...this project is probably not a good starting point to learn coding - on the opposite, for experienced coders it is no problem to read ...so I don't think it is a good idea to expect this extra work from him...

uholeschak, 2022-07-12:
Basically the idea is, that the code is self explaining.
I think that comments are only required in special situations where this is not obvious.
In moste cases where there are comments there are outdated ...

closed 2022-07-12 by uholeschak