uholeschak/ediabaslib #130: Cannot set up the app

Bmwe39528, 2022-10-03:
I tried installing the app multiple times, tried beta app as well. When the app warns me that it will extract 2.4GB if files, I click "ok" and the whole process only takes two minutes. I don't know how it's possible to download 2.4GB data this fast, or if it doesn't download for some reason.
I cannot find the downloaded 2.4GB data.

I tried following the instructions on the screenshot from https://xtrons.ibus-app.de/index.php?title=DeepOBD_-_Adapter_und_App_einrichten#Zusammenfassung_und_Download but cannot find this folder anywhere
Android/data/de.holeschak.bmw_deep_obd/files

I will include some pictures and the instructions I follow. I'm not sure what I'm doing wrong?
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

uholeschak, 2022-10-03:
If you don't have direct access to the app folder you have to use the "File management" options of the global settings to copy the external directory to the app:
https://github.com/uholeschak/ediabaslib/blob/master/docs/GlobalSettings.md

closed 2022-10-03 by uholeschak

uholeschak, 2022-10-03:
Officially it's recommended to use the configuration generator and not external files.
If there are problems with external files contact the supplier of these files.