uholeschak/ediabaslib #132: inpa D-CAN/K-line protocol and firmware

gauix4, 2022-10-12:
Hello
After several weeks of research I can not find the following information
which protocol uses INPA on a mini R50/53 because it is much faster than ISO 9141
I am also looking for the open source firmware of the D-CAN/K-Line cable because I find .bin which is absolutely not open source

I also tried to find threads or forums there is nothing at all
I'm looking to get myself a dashboard so no pc in the car but an ESP32
a computer has no place in a car that's why I'm looking to do everything on a microcontroller with a screen to view data only there is no information on the internet on k-line communication on mini or BMW
so i can't start any projects until i know what communication it uses moreover there is no open source firmware findable for inpa usb converter every time it's binary files so it's not Open Source I don't know why people call it open source!

I am therefore sending you this message hoping that you have answers to these questions your project is beautiful but for my part they use a computer which I do not want there are already at least the IDs that I can recover in your code this who is already good
thank you very much !

uholeschak, 2022-10-12:
Sorry I have no experiences with Mini.

closed 2022-10-12 by uholeschak

gauix4, 2022-10-12:
> Désolé, je n'ai aucune expérience avec Mini.

there is not only mini all the vehicles which are on INPA use the same protocol I do not find which protocol it uses
you have encoded the IDs so you must have an idea of what protocol it is to have them encoded
```cpp
private static string Type104Convert(UdsReader udsReader, byte[] data)
        {
            if (data.Length < 6 + 1)
            {
                return string.Empty;
            }

            byte maskData = data[0];
            StringBuilder sb = new StringBuilder();
            for (int i = 0; i < 3; i++)
            {
                StringBuilder sbValue = new StringBuilder();
                bool bData1 = (maskData & (1 << i)) != 0;
                bool bData2 = (maskData & (1 << (i + 3))) != 0;
                if (bData1 || bData2)
                {
                    sbValue.Append("IAT ");
                    if (bData1)
                    {
                        sbValue.Append($"1{i + 1}");
                    }
                    if (bData2)
                    {
                        if (bData1)
                        {
                            sbValue.Append("/");
                        }
                        sbValue.Append($"2{i + 1}");
                    }
                    sbValue.Append(": ");

                    if (bData1)
                    {
                        byte value = data[i + 1];
                        double displayValue = value - 40.0;
                        sbValue.Append($"{displayValue:0.}");
                    }
                    if (bData2)
                    {
                        byte value = data[i + 3 + 1];
                        double displayValue = value - 40.0;
                        if (bData1)
                        {
                            sbValue.Append("/");
                        }
                        sbValue.Append($"{displayValue:0.}");
                    }
                    sbValue.Append(" ");
                    sbValue.Append(GetUnitMapText(udsReader, 3) ?? string.Empty); // °C

                    if (sb.Length > 0)
                    {
                        sb.Append("; ");
                    }
                    sb.Append(sbValue);
                }
            }

            return sb.ToString();
        }
```

uholeschak, 2022-10-12:
This function is for VAG sorry.

uholeschak, 2022-10-12:
I would recommend to use ISTA-D and log the communication.

gauix4, 2022-10-12:
I'll look into that thanks but I think it still uses ISO 9141 for diagnostics anyway

uholeschak, 2022-10-12:
The protocol depends for the ecu file used, so you have to check the corresponding ecu file.
Ediabas only gets the parameters.

gauix4, 2023-02-09:
Where is this file stored ? Please :) 
I can't find when the files each time I come across the VAG

My ECU is EMS2K
Siemens C167

uholeschak, 2023-02-09:
What exactly do you wan to know?
Simply check the apps ecu directory where the files are extracted.

gauix4, 2023-02-09:
I'm looking for the UDS to decode this
its data ans generated and retrieved by ediabas so it is probably in the files you have attached but I can't find them there are too many files I tested with the search function it only gives me VAG

[image]

uholeschak, 2023-02-09:
I think analyzing the ecu file will not really help you, its too complicated.
Better to simply reverse engineer form the logged data you have.

gauix4, 2023-02-09:
ediabas the deciphers
I will therefore have to find my answer in ediabase normally

It's to code all this on an ESP32
I can already talk to the ECU and I understand a lot of the data

uholeschak, 2023-02-09:
I expect that I also simply don't know the answer, the info probably in hidden in manufacturer specific files.

gauix4, 2023-02-09:
EdiabasLib/UdsFileReader/UdsReader.cs
I'm not sure this and that for wv if ?

uholeschak, 2023-02-09:
Yes this is for VAG. But the manufacturer specific data is not included in the repo and is encrypted to prevent extraction.
Nobody will help you to extract this data.