uholeschak/ediabaslib #133: Improve zip file extracting

mi-hol, 2022-10-15:
[image]

The current logic should be improved to:

1. check available free space on configured storage device
2. read "expanded" file size from zip archive
3. compare the 2 sizes and only if not enough free is available
4. display error with the exact amount of space missing, abort extraction before free space is exhausted!

uholeschak, 2022-10-15:
Basically in the current beta version there is check for the free space that should be correct.
This message is only  fallback if something unknown happens.

closed 2022-10-15 by uholeschak

mi-hol, 2022-10-15:
> This message is only fallback if something unknown happens.
> I would expect that there is a problem in the file system layer.

then text of message should reflect that

uholeschak, 2022-10-16:
I could remove the sentence: "Is there no space left on the device?"
The exception message is below in your screenhsot.
Sharing violation. I expect another software accesses the file at the same time in the background.

mi-hol, 2022-10-16:
> Sharing violation. I expect another software accesses the file at the same time in the background.

Installers for Windows handle this situation by:
1. retry after a few seconds
2. if it still fails, the schedule replacing file at next boot.
Would that work for Android too?

> I could replace the sentence: "Is there no space left on the device?" with "Is the file system corrupted?"

Thanks. 

Looking further I wonder if the error text just below the change https://github.com/uholeschak/ediabaslib/commit/9cdbe53c1bbde1793846efccc96c45f1f2adc0e0#diff-edd0e096b54aaced05b92f2752582621b5806679d949823f0e02925fbad9d276R152
should get the same or similar change?

uholeschak, 2022-10-16:
There is already special handing for protection software file locks.
But these are many files and this is simply too much for most protection software. This is why there is the hint to remove the software.
Yes the text below is the change.