uholeschak/ediabaslib #136: CanAdapter Firmware Ignition / Clamp-Handling not working

DaniM54, 2022-11-06:
Hey,

I have problems with the ignition detection. I'm using a standard K+DCAN interface on an E46. With the original firmware, clamp handling works without any problems. As soon as I flash the combined.hex, it no longer works (does not appear).
When I flash it back it works again with no problems.

Does clamp handling only work with CAN via the 0x130 message, or should it also work "analogue" with older systems? I took a quick look at the code, but unfortunately I don't have the know-how to understand it :(

And another question... when I compile the CanAdapter project, I get the CanAdapter.hex file. But how do I get the combined.hex including the bootloader for flashing via USB?

Thank you for your immense efforts on this project!

uholeschak, 2022-11-06:
Sorry I can't remember any more and this part of the project is old und unsupported.
Maybe your hardware has different layout and other pins for the signals are used.

closed 2022-11-06 by uholeschak

DaniM54, 2022-11-06:
Ok, thank you anyway for the fast reply.

For documentation:
I have tried to understand various things on my hardware and the ignition recognition. With the original firmware of the interface there is actually +5V on PB4 when the ignition is on. Interestingly, this is not the case with the modified firmware (which also points to PB4). I can't explain why at the moment.
Normally there is about 2.5V on PB3, and then 5V with the ignition. This seems to be enough for the ignition detection to work if I replace PB4 with PB3 in the code. (With a little delay, but sufficient in practice.)

After compiling I copied the bootloader out of the old Combined.hex and attached it to my newly compiled CanAdapter.hex, that worked and I can flash the chip via USB without any problems.

My problems are solved, maybe it will help someone else.

Regards, Daniel

uholeschak, 2022-11-06:
Thanks, so maybe your hardware is incompatible ...
The combined file is created in a post build event of the project:
`
del $(OutputDirectory)\Combined.hex
IF EXIST $(MSBuildProjectDirectory)\..\fastboot\bootload.hex (
type $(OutputDirectory)\$(OutputFileName).hex | findstr /v :00000001FF > $(OutputDirectory)\Combined.hex
type $(MSBuildProjectDirectory)\..\fastboot\bootload.hex >> $(OutputDirectory)\Combined.hex
)
`
So this part seems to not work in your case.