uholeschak/ediabaslib #137: Problem with ELM327 adapter.

shar34, 2022-12-01:
I flashed **PIC18F25K80** with **CanAdapterElm.X.production.unified.hex** from **default** folder. Flashing successful:

```
Connecting to MPLAB PICkit 3... 

Currently loaded firmware on PICkit 3
Firmware Suite Version.....01.56.09
Firmware type..............PIC18F
Programmer to target power is enabled - VDD = 5.000000 volts.
Target device PIC18F25K80 found.
Device Revision ID = 6
Loading code from /home/user/Downloads/Binaries-20220213/CanAdapterElm/default/CanAdapterElm.X.production.unified.hex...
2022-12-01 22:36:51 +0300 - Hex file loaded successfully.
2022-12-01 22:36:53 +0300 - Programming...

Device Erased...

Programming...

The following memory area(s) will be programmed:
program memory: start address = 0x0, end address = 0x7fff
configuration memory
Programming/Verify complete
2022-12-01 22:37:02 +0300 - Programming complete.
```

Then I try send `82 F1 F1 FE FE 60` command and receive valid response. **Deep OBD** show **0.15** firmware, but when I try connect to auto I get error, I try reflash **ELM327 firmware** over air and adapter get bricked state with all led lights with error in **Deep OBD**.

Then I try flash ELM327 firmware with **PICKIT3** from **default** folder, **Deep OBD** recognized that adapter in **ELM327 mode** and suggest flash alternate firmware, then adapter, like first case, also get bricked.

[image]

uholeschak, 2022-12-01:
This only means that the firmware updated failed and the device is boot loader mode. Probably a Bluetooth a communication problem.

closed 2022-12-01 by uholeschak

uholeschak, 2022-12-01:
I have tested it gain, no problem. I expect your Bluetooth chip has incompatible timing parameters.