uholeschak/ediabaslib #140: Connect to ELM327 Deep OBD WiFi (ESP8266) with STA mode.

shar34, 2023-01-23:
I create WiFi access point on Android device and connect ESP8266 adapter to this network. I can get web interface, but in Deep OBD application there is no IP address manual setting:
[image]
[image]

If I connect Android device directly to ESP8266 WiFi point, buttons "read" and "device configurations" become available. Is possible set manual IP address (192.168.236.154 like my photo for example) when Android device in WiFi access point mode?

uholeschak, 2023-01-23:
At the moment I am working to set the IP and port in the GUI manually. Then you could set any IP address for the adapter you want.

shar34, 2023-01-23:
I'll be waiting, thank you very much!

closed 2023-01-23 by shar34

uholeschak, 2023-01-28:
In the current Beta there is the possibility to edit the adapter IP address and port in the menu. In Hotspot mode (AP on, WiFi off), you must enter the adapter assigned IP address and port otherwise communication will not start.
There is no official support in Android for Hotspot mode in the app, so there is no guarantee that this will work.

shar34, 2023-01-28:
Tried on Android multimedia headunit, everything works well, thanks!
[image]

Maybe set port to **23** by default (in ELM327 Deep OBD WiFi ESP8266EX manual device configuration)?

uholeschak, 2023-01-28:
Thanks for the info. You mean in Hotspot mode the default port is 23? Are there other WiFi devices that could be used in Hotspot mode that have a different port?

shar34, 2023-01-29:
ESP-Link use 23 port by default in serial bridge: `In order to connect through the esp-link to a microcontroller use port 23.`
[image]

uholeschak, 2023-01-29:
Yes I know, but the IP edit is also intended for other WiFi adapters in non Hotspot mode. All other adapters use port 35000. 
What I could perhaps do: If the port is missing and Hotspot mode is active, the default port would switch to 23, because only ESP is able to work in Hotspot mode.

uholeschak, 2023-01-29:
In the next beta the default port in hotspot mode is 23 (if not specified).