uholeschak/ediabaslib #141: PRG/GRP-Dateien nach Text durchsuchen

akelsch, 2023-02-03:
Hallo Ulrich,

ich bin Mal so frei auf Deutsch zu schreiben.

Ich bin dir für deine App ziemlich dankbar, da ich damit nun einen Batteriewechsel registrieren und nachprüfen kann. Nur hat es ziemlich lange gedauert, bis ich die entsprechenden Jobs herausgefunden habe (`STATUS_SYSTEMCHECK_PM_INFO_1` und `STEUERN_BATTERIETAUSCH_REGISTRIEREN`).

Mir ist bewusst, dass sich das Tool als solches für Experten versteht. Dennoch würde ich gerne fragen, ob man in den PRG/GRP-Dateien irgendwie nach Texten suchen kann. In der App ist das leider viel zu umständlich, da ich jeden Job einzeln öffnen müsste um seine Parameter zu betrachten. In meinem Fall habe ich (vor allem ersteren der beiden Jobs) erst in einem [Forum-Beitrag](https://www.e90post.com/forums/showthread.php?t=1467274) mit angehängtem Text-Dump gefunden. Davor hatte ich ohne Erfolg versucht, die von der App heruntergeladene Datei (msd80n43.prg) im Hex-Dump anzuschauen oder in den "EcuFunctions" (XML) zu suchen.

Vielleicht gibt es da ja eine bessere Möglichkeit.

Danke!

uholeschak, 2023-02-03:
Die Staus Jobs auf der ECU Seite kann man oben filtern, allerdings nur nach dem Job Namen.
Im Ediabas Tool gibt es keine Filter, die Jobs sind alphabetisch sortiert. wenn ich das richtig verstehe, hilft aber wohl ein Filter nach dem Job Namen auch nicht. Wie ist das den in Tool32? Vermutlich haben sie dort das gleiche Problem.
Tool32 ist nur dafür gedacht Jobs auszuführen, bei denen man weiß was sie tun. Der Text kann ja auch mal DE oder EN sein.
Eigentlich ist der Weg über Google korrekt.

akelsch, 2023-02-03:
Ja, in meinem Fall hätte das wohl nicht sonderlich viel geholfen, da mein Suchbegriff "BATTERIETAUSCH" war und ich diesen letztendlich in dem Ergebnis `STAT_BATTERIETAUSCH_LETZTER_WERT` von besagtem Job gefunden habe. Das Internet hat mich hier zuerst leider auf die falsche Fährte geführt, da dort der Job `STATUS_BATTERIE_HISTORY` genannt wurde, dieser aber bei meinem Wagen nicht zu finden war.

Wenn man sich Tool32 in Videos anschaut, scheint es auch nur eine einfache Auflistung zu sein. Ich dachte nur, da die App sowieso die Texte zu Jobs und Ergebnissen anzeigt, dass man diese vielleicht so etwas wie "entpacken" könnte, um solch eine Textdatei wie in dem verlinkten Forumsbeitrag zu erhalten. Aber da bleibt wohl nur Googlen oder im Zweifel durchklicken.

uholeschak, 2023-02-03:
Aber `STEUERN_BATTERIETAUSCH_REGISTRIEREN` hätten sie doch damit gefunden oder nicht? Wenn man nach Batterie sucht? Das Ganze in eine Textdatei zu exportieren ist wohl auch nicht so richtig hilfreich, die müsste man sich wohl dann erst auf dem PC übertragen

akelsch, 2023-02-03:
Diesen Job hätte man in der Tat mit der Suche der App finden können. Allerdings war das in diesem Fall nicht nötig, da der Job in gängigen Blogs und Foren sowieso geläufig war und über verschiedene Modelle hinweg vom Namen her identisch geblieben zu sein scheint. Mit dem zweiten Job hatte ich dann meine Probleme, da dieser am Beispiel eines E65 verwendet wurde, ich diesen aber beim E92 nicht finden konnte (wobei die Ergebnisse teilweise identisch heißen, was mir dann bei der Suche geholfen hat).

Wenn man alle Strings direkt in der App durchsuchen könnte, wäre das natürlich am praktischsten 😄

uholeschak, 2023-02-12:
I have now uploaded a new beta with general improved search functions.

closed 2023-02-12 by uholeschak