uholeschak/ediabaslib #144: Deep Obd & OBD adapter

ati1977, 2023-02-16:
Hi, can you please advice me which adapter I can use with deep obd soft on my car headunit with  android?
My car is 2006 BMW E83 3.0 diesel. For this moment I use rheingold with USB K-DCAN (with not sure which but think 7 and 8 pin connected together) adapter and it is also works with deep obd.
But I want BT adapter and afraid that can buy a wrong.
For this moment is interesting for me this adapters:
[http://www.vgatemall.com/products-detail/i-11/](https://4pda.to/pages/go/?u=http%3A%2F%2Fwww.vgatemall.com%2Fproducts-detail%2Fi-11%2F&e=120893634)
[http://www.vgatemall.com/products-detail/i-9/](https://4pda.to/pages/go/?u=http%3A%2F%2Fwww.vgatemall.com%2Fproducts-detail%2Fi-9%2F&e=120893634)
[https://xtrons.com/obd…-tool-obd02#review-sec](https://4pda.to/pages/go/?u=https%3A%2F%2Fxtrons.com%2Fobd2-ii-scanner-tool-obd02%23review-sec&e=120798649)
[https://xtrons.com/obd-ii-scanner-tool-obd01s](https://4pda.to/pages/go/?u=https%3A%2F%2Fxtrons.com%2Fobd-ii-scanner-tool-obd01s&e=120798649)
If this adapters will not work in my case, can you advice me which one will work.
Thank you very much for support

closed 2023-02-16 by uholeschak

ati1977, 2023-02-16:
Thank you very much for support and explanation. Unfortunately for this moment 42euro with shipping is very high price for my budget. Will try first with my USB cable. Thank you, have a nice day.

uholeschak, 2023-02-16:
The component price has increased very much in the last time.

ati1977, 2023-02-16:
Sure I understand you. I am not from EU and for our currency it is other price. I will buy this adapter but later. Thank you
PS Which one I need BC417 or YC1021? And what is a difference?
Thank's a lot for support

uholeschak, 2023-02-16:
BC417 has Bluetooth protocol 2.1 and YC1021 protocol 3.0. For newer Android devices YC1021 is ok.

ati1977, 2023-02-16:
Ok. Thank you again for such good support.

*With Best Regards*

*Armen Sarkisyan*

ati1977, 2023-02-17:
Hi again, can I ask you another one question :) 
Can I use your adapter BC417 or YC1021 with 7+8 pin bridge to scan other brand cars with different software?
Thank you

uholeschak, 2023-02-17:
The Deep OBD app supports also VAG vehicles until 2017-08.
But could also flash the ELM327 firmware to the adapter via the Deep OBD app, then you could use it with other ELM compatible apps.

ati1977, 2023-02-17:
Ok. So if I will flash the ELM327 firmware to the adapter via the Deep OBD app, then I can use this adapter  for example in Nissan cars and for example torque app? Thank you

uholeschak, 2023-02-17:
If you find an ELM327 compatible app that support Nissan yes.

ati1977, 2023-02-17:
Ok, Thank you very much Also is this adapter your or it is different and not suitable for me?
https://shop.ibus-app.de/i-bus-app/zubehoer/interfaces-module/258/deep-obd-obd-ii-bluetooth-adapter
PS Minus of this adapter is that it does not have power button and not going to sleep after 30 minutes

ati1977, 2023-02-17:
Ok. Thank you. Great work and it is very good that such professionals like you makes so good things for users like me and other people. Good luck and best wishes. Thank you

ati1977, 2023-02-18:
Hi again, can I ask you about canbus maybe you have answer. Is it possible to update canbus on my head unit or it is only possible with firmware from the company which made this canbus& I have this one on my BMW E83 2006 (not restyling) 3.0 d 218hp. This is photo of my canbus. Thank you
[image]

uholeschak, 2023-02-18:
Sorry I don't know this item, ask the manufacture.